RocketPy-Team/RocketPy #27: Rail Buttons Force and Moments Calulation

giovaniceotto, 2019-02-10:
There is no current way to determine the forces and moments on the pair of rail buttons which guide High Powered Rockets in the launch rail.

This could probably be implemented in the post processing phase of the flight simulation, in the `Flight.postProcess` method.

Gui-FernandesBR, 2022-01-31:
Currently there are evaluations for Normal and Shear Forces on Rail Buttons,
It's lacking only bending moments calculations.
Altought I don't see too many uses of such values, I believe providing it to final user will be fast, simple, and beautiful. Then I'm marking it as a good first issue.

ps: Perhaps in the future we can create a place on rocktpy to be dedicated to structural analysis only.

Gui-FernandesBR, 2024-02-12:
Quoting @giovaniceotto here:

Everything depends on the assumptions made about the forces and moments exerted by rail buttons. By comparing the outcomes of uDot and uDotRail, it becomes apparent which forces and moments are necessary for the rocket to stay on its track (while on the launch rail).
The challenge arises when attempting to divide this into two Rail Buttons, assuming they can apply any force and moment, leading to a statically indeterminate system.
However, a potential solution by considering that the rail buttons exert force only perpendicular to the track, indicating that in most cases, there might be a viable solution, although some situations may still result in system indefiniteness, likely limited to specific singularities.