RocketPy-Team/RocketPy #28: Simulate Single Rail Button Flight Phase

giovaniceotto, 2019-02-18:
The current Flight simulation does not take into account the fact the for a very short amount of time, a high powered rocket is guided through the launch rail with a single rail button, after the first has already cleared the rail. Under this condition, the rocket displays a planar motion, with 3 degrees of freedom.

This should be implemented as a new flight phase due to its importance to dynamic stability when the rocket is clearing the rail.

giovaniceotto, 2019-02-18:
Working on it in the [rail_buttons](https://github.com/giovaniceotto/RocketPy/tree/rail_buttons) branch.

Gui-FernandesBR, 2026-11-02:
Reopened: the fix is merged on `develop` and will close when v1.14.0 is released to `master`.