RocketPy-Team/RocketPy #31: Recovery options and drag coeefficients?

concretedog, 2020-06-12:
Hey, love this project! I'm the founder of open research rocketry and i'd love to chat to you about various aspects of this and some of our burgeoning projects over in that community. Was wondering if you had/would consider adding different recovery system design options (we have some parachute design calculators and scripts over on our webpage and gitlab repo). We are using either a designed diameter or a reference surface area and a drag coefficient and I wondered if it could be built into your project? (Full disclosure... I am much more mechanical so I might be way off the mark!) If you are on riot/matrix we would love to have you drop in the chat! https://riot.im/app/#/room/#OpenResearchRocketry:matrix.org

giovaniceotto, 2020-07-24:
@concretedog That seems like a great idea!

Currently, RocketPy allows the user to enter the product Cd*S, where Cd is the effective drag coefficient and S is the reference area used. However, we currently do not have any available methods to calculate Cd for the user.

I believe it's important to always have the flexibility of letting the user enter the Cd*S value directly, especially when the user has the precise value measured.

That being said, I completely agree we could have additional specialized functions to calculate or even use tabulated values of Cd for different parachute types. Having approximated terminal velocity as input instead of Cd*S would be more practical.

I searched a little bit through Open Research Rocketry's website and GitLab. and found really interesting materials! Congratulations for the job you guys are doing! More specifically, I saw a couple of scripts which helped on determining the parachute's geometrical parameter, which would also be nice to integrate with RocketPy. Let me know what you think! 

Unfortunately, I could not find anything regarding how to get the values of Cd. Do you have any resources to help here?

Gui-FernandesBR, 2021-09-20:
@FranzYuri

Gui-FernandesBR, 2022-02-13:
The class parachute is coming!! #113

Gui-FernandesBR, 2024-08-19:
Logan Herrera [ESRA] — 12/15/2022 1:32 PM

> I just remembered I myself implemented all the functions in T W Knacke’s para
> chute book when I was a student
> It’s in MATLAB but still
> You are welcome to include it
> Also gives opening time and distance
> https://github.com/stanford-ssi/rockets-irec/tree/master/Recovery/Parachute%20Design%20and%20Analysis