RocketPy-Team/RocketPy #34: Rail Departure Velocity and time

Gui-FernandesBR, 2020-07-28:
From what I've known for so long using RocketPy (you can correct me if necessary), the rail departure time is assumed to be the time when the Rocket CM crosses the top of the Launch Rail. However, the main international rocket competition (SACup) defines rail departure time as "the first instant in which the launch vehicle becomes free to move about the pitch, yaw, or roll axis."[[DTEG - SACUP](http://www.soundingrocket.org/uploads/9/0/6/4/9064598/sa_cup_irec-design_test___evaluation_guide_20191118__rev_c_final_.pdf)]

I believe that we must make some changes to the Flight class so that the railDepartureTime variable be considered as the time that the upper button crosses the top of the rail.

giovaniceotto, 2020-07-28:
I agree with you completely in this one, @Gui-FernandesBR. Initially, I wanted to implement a set of new equations of motion which described the flight of the rocket with a single rail button still sliding along the rail.

However, the equations of motion have proven quite challenging and this will take a while to correctly derive, implement and test.

For the meantime, do you have any suggestions on how to mitigate this issue?

Gui-FernandesBR, 2020-07-28:
I made a pull request a few minutes ago and I believe the problem was already solved, at least until we implement a set of new equations of motion.

I created the 'effective1RL' attribute to be used as the new Rail length in all simulations. It discounts the distance between the nozzle exit and the upper button. However, the original rail length inserted by the engineer is still saved in another attribute.

Here's a copy of the script created:
    `        upperRButton = max(self.rocket.railButtons[0])`
    `        lowerRButton = min(self.rocket.railButtons[0])`
    `        nozzle = self.rocket.distanceRocketNozzle`
    `        self.effective1RL = self.env.rL - abs(nozzle - upperRButton)`
    `        self.effective2RL = self.env.rL - abs(nozzle - lowerRButton)`

giovaniceotto, 2020-07-28:
Just for information purposes: the pull request being cited is #36.

giovaniceotto, 2020-07-28:
@Gui-FernandesBR, did you replace all occurrences of `self.env.rL` with `self.effective2RL` in the Flight class? Or just the ones used to compute rail departure conditions?

By the way, perhaps `outOfRailXxxxxx ` should be renamed to `railDepartureXxxxxx`.

giovaniceotto, 2020-08-01:
I will close this issue since there is already another one about implementing the rest of code needed: #28.

closed 2020-08-01 by giovaniceotto