RocketPy-Team/RocketPy #43: Can you tell me how to use the program to modify the trajectory of the rocket, for example, the second half can automatically turn like an airplane

1604010326, 2020-09-03:
Can you tell me how to use the program to modify the trajectory of the rocket, for example, the second half can automatically turn like an airplane

giovaniceotto, 2020-09-03:
Sure!

Let me make sure I understood you correctly. You would like to control the rocket by applying custom forces to it?

1604010326, 2020-09-04:
> 当然！
> 
> 让我确保我正确理解了您。您想通过向其施加定制力量来控制火箭吗？

Yes, at present, the rocket is only raising and landing. I want him to do a horizontal flight after rising and then land, but I don't know what to do

1604010326, 2020-09-04:
Or other shapes of the trajectory, in the second half of the flight can also control it to do horizontal or turning action, which part of the program should I modify

giovaniceotto, 2020-09-04:
It seems like you maybe launching the rocket vertically with no wind.

You can add wind in the Environment class through the `setAtmosphericModel` method.

To change the inclination of the launch rail, you can set the input `inclination` when initializing the Flight class.

Could you share your code so that I can help you better?

giovaniceotto, 2020-09-04:
In terms of control, you would have to modify the method `Flight.uDot`.

It computes the forces R1, R2 and R3 and the moments M1, M2 and M3 during flight.

You could add additional terms to these forces and moments, essentially the effects of you control system.

1604010326, 2020-09-04:
> 在控制方面，您必须修改method `Flight.uDot`。
> 
> 它计算飞行过程中的力R1，R2和R3以及力矩M1，M2和M3。
> 
> 您可以在这些力和力矩上添加其他术语，实质上是控制系统的作用。
> 
> 让我知道这是否是您需要的。如果可以的话，我可以进一步帮助您。

First of all, thank you. At present, I only successfully run the sample code you gave. I want to do a simulation similar to missile trajectory, but your project is rocket simulation. Do you think the project can simulate missile trajectory

1604010326, 2020-09-04:
If you can, would you please tell me how to do it, or can you provide a sample program? Thank you very much

1604010326, 2020-09-04:
If the rocket is launched horizontally at the beginning, how to set the angle? I set it to 0. If I want it to launch horizontally, it will report an error。
my code
TestFlight = Flight(rocket=Calisto, environment=Env, inclination=0, heading=0)

giovaniceotto, 2020-09-04:
Indeed launching with `inclination=0` can be quite problematic, since the code things the rocket impacts the ground immediately. 

I would suggest running with `inclination=3` or above. It seems to work well for such value and the rocket maximum altitude is below 1 meter.

giovaniceotto, 2020-09-04:
RocketPy is indeed focused on rockets simulation. However, I do believe you might me able to use it for your purposes with small changes.

If the missile like craft you want to simulate does not have active control, I don't think any changes are necessary.

If active control is needed, then you will need to modify `Flight.uDot` and add a couple of lines there to calculate the control forces on your craft, as a function of the state variables. Does this make sense?

1604010326, 2020-09-04:
> 确实，用火箭发射`inclination=0`可能会造成很大的问题，因为火箭的代码会立即影响地面。
> 
> 我建议跑`inclination=3`或以上。对于这样的值似乎效果很好，并且火箭的最大高度在1米以下。但是，您无法使用任何降落伞来工作。因此，请删除所有定义降落伞的行，也就是`Callisto.addParachute`。

OK, thank you so much. I'll try to modify it. Can I change the trajectory of the second half of the rocket by modifying the transmitter part? If it's not active control, I guess if you want to modify the second half of the trajectory, you can only modify the wind, right?

1604010326, 2020-09-04:
> Indeed launching with `inclination=0` can be quite problematic, since the code things the rocket impacts the ground immediately.
> 
> I would suggest running with `inclination=3` or above. It seems to work well for such value and the rocket maximum altitude is below 1 meter. However, you cannot have any parachutes for this to work. So remove all lines which defines parachute, a.k.a `Callisto.addParachute`.

OK, thank you so much. I'll try to modify it. Can I change the trajectory of the second half of the rocket by modifying the transmitter part? If it's not active control, I guess if you want to modify the second half of the trajectory, you can only modify the wind, right?

giovaniceotto, 2020-09-04:
Sorry, but what do you mean by second half of the rocket? Is it a two stage rocket?

1604010326, 2020-09-04:
> RocketPy is indeed focused on rockets simulation. However, I do believe you might me able to use it for your purposes with small changes.
> 
> If the missile like craft you want to simulate does not have active control, I don't think any changes are necessary.
> 
> If active control is needed, then you will need to modify `Flight.uDot` and add a couple of lines there to calculate the control forces on your craft, as a function of the state variables. Does this make sense?
> 
> If you can describe your problem with more details, perhaps I could even help with new code implementation!

I want to send a picture to show you the track I want more clearly, but I don't know how to send it out? I mainly want to simulate the trajectory, such as flying horizontally first, seeing the target, then flying (like a rocket) and then landing to hit the target

1604010326, 2020-09-04:
> Sorry, but what do you mean by second half of the rocket? Is it a two stage rocket?

The second half of a flight, such as the landing phase

1604010326, 2020-09-04:
> > Sorry, but what do you mean by second half of the rocket? Is it a two stage rocket?
> 
> The second half of a flight, such as the landing phase

Generally speaking, my question is, for example, when the rocket rises to the highest point, I don't want it to land directly, but fly horizontally for a period of time and then land. What should I do?

1604010326, 2020-09-04:
Hello, I see that your program can apply the wind of X axis and Y axis(wind_u,wind_v)

Is there any wind on the Z axis? for example wind_w?

giovaniceotto, 2020-09-05:
Yup, we do not have wind_w because it is generally not relevant for rockets.

You can make the rocket climb after reaching apogee, for sure, however, you need propulsion for that! With out any propulsion you will just fall.

It seems to me that all you need is a controlled propulsion, such as TVC and perhaps bigger fins/wings. Do you agree?

1604010326, 2020-09-05:
I quite agree with you. As you said, if I want to add a propulsion force at a certain position or time when the rocket lands, how should I modify it in the code? In addition, do you have any references or formulas for calculating the forces and trajectories of the rocket? I would appreciate it if you could describe it in detail

giovaniceotto, 2020-09-06:
Lets say you want to add a certain propulsion force to the rocket. This means you want to give the values of:

- CF1
- CF2
- CF3

Which are the 3 components of control force that act at the center of mass of the rocket, in Newtons. You may also want to specify the control moments:

- CM1
- CM2
- CM3

Which need to be given in Newton meters. That being said, you need some information to determine these forces and moments. The information which can be easily made available is the time and state of the rocket:

- t: time since launch in seconds
- x, y, z: rocket coordinates in meters
- vx, vy, vz: velocity components in m/s
- e0, e1, e2, e3: euler parameters or quaternions
- w1, w2, w3: angular velocity components in rad/s

In term of code implementation, you could modify the function `uDot` in `rocketpy/Flight.py` by adding the following lines right before line [1337](https://github.com/giovaniceotto/RocketPy/blob/6410eefedfe002badb2e0b3660ae163b6b9e3527/rocketpy/Flight.py#L1337):

```python

CF1, CF2, CF3, CM1, CM2, CM3 = your_costum_propulsion_function(t, x, y, z, vx, vy, vz, e0, e1, e2, e3, w1, w2, w3) 
R1 += CF1
R2 += CF2
R3 += CF3
M1 += CM1
M2 += CM2
M3 += CM3

```

This should handle control forces which can be given without significant changes to the rocket mass. Does that work for you?

1604010326, 2020-09-06:
> Lets say you want to add a certain propulsion force to the rocket. This means you want to give the values of:
> 
> * CF1
> * CF2
> * CF3
> 
> Which are the 3 components of control force that act at the center of mass of the rocket, in Newtons. You may also want to specify the control moments:
> 
> * CM1
> * CM2
> * CM3
> 
> Which need to be given in Newton meters. That being said, you need some information to determine these forces and moments. The information which can be easily made available is the time and state of the rocket:
> 
> * t: time since launch in seconds
> * x, y, z: rocket coordinates in meters
> * vx, vy, vz: velocity components in m/s
> * e0, e1, e2, e3: euler parameters or quaternions
> * omega1, omega2, omega3: angular velocity components in rad/s
> 
> In term of code implementation, you could modify the function `uDot` in `rocketpy/Flight.py` by adding the following lines right before line [1337](https://github.com/giovaniceotto/RocketPy/blob/6410eefedfe002badb2e0b3660ae163b6b9e3527/rocketpy/Flight.py#L1337):
> 
> ```python
> CF1, CF2, CF3, CM1, CM2, CM3 = your_costum_propulsion_function(t, x, y, z, vx, vy, vz, e0, e1, e2, e3, omega1, omega2, omega3) 
> R1 += CF1
> R2 += CF2
> R3 += CF3
> M1 += CM1
> M2 += CM2
> M3 += CM3
> ```
> 
> This should handle control forces which can be given without significant changes to the rocket mass. Does that work for you?

Thank you for your guidance. I will try it in the near future

1604010326, 2020-09-06:
> Thank you for your guidance. I will try it in the near future

I have tried the following preliminary results, but I don't know which direction of power  R1, R2 and R3 represent respectively. Can you give me a hint？

giovaniceotto, 2020-09-07:
Sure! So `R3` is directly aligned with the rocket's axis of symmetry. Take a look at this image:

[image]

`R3` would be the `z` axis, `R1` and `R2` are `x` and `y` respectively.

1604010326, 2020-09-08:
> Sure! So `R3` is directly aligned with the rocket's axis of symmetry. Take a look at this image:
> 
> [image]
> 
> `R3` would be the `z` axis, `R1` and `R2` are `x` and `y` respectively.

Thank you very much. Your reply is very helpful to me. I have been able to change the flight direction of the rocket in the later stage of its flight. But I want to make the rocket fly smoothly for a period of time. I try to test different forces, but it is difficult to find the right force to keep the rocket flying horizontally for a period of time. Do you have any suggestions?

giovaniceotto, 2020-09-08:
I imagine the only good way of making sure the rocket follows a trajectory you desire is by implementing a path-following control algorithm. Take a look at:

- [https://link.springer.com/chapter/10.1007/978-3-642-38253-6_6](https://link.springer.com/chapter/10.1007/978-3-642-38253-6_6)
- [https://www.researchgate.net/publication/232042027_A_Path-Following_Algorithm_for_Missiles](https://www.researchgate.net/publication/232042027_A_Path-Following_Algorithm_for_Missiles)

1604010326, 2020-09-10:
> I imagine the only good way of making sure the rocket follows a trajectory you desire is by implementing a path-following control algorithm. Take a look at:
>
> * [...]

OK, thank you very much. I'll take a look

Gui-FernandesBR, 2022-01-31:
Hey, I'm closing this issue once I believe problems form user @1604010326 were solved.

However, feel free to re-open in case you have any other questions regarding that ;)

closed 2022-01-31 by Gui-FernandesBR