RocketPy-Team/RocketPy #46: Alpha and Beta for Cd assumptions?

nbrophy3, 2020-09-21:
I noticed that your Cd is purely from (I assume Rasaero) Power on and Power off cd data. With Missile Datcom readers and other simulations, either Alpha or both Alpha and Beta values are plotted along with Mach in order to yield a Cd at a certain moment in time. Do you make any adjustments to your Cd in order to account for this difference (like some kind of offset)? Or does your code make the assumption of small angle of attack and symmetric about an axis to avoid using this information? It's a good assumption tbh, if that's what you're doing, but how do your results compare to something like Rasaero or Open Rocket?

giovaniceotto, 2020-09-23:
@Gui-FernandesBR

Gui-FernandesBR, 2020-09-29:
I'm sorry @giovaniceotto, but you are probably the best person to answer that question. But I will try...

@nbrophy3 , in fact, Rocketpy considers the drag force (R1) as always parallel to the rocket's axis. We know that the most common literature assumes that the drag acts in a direction opposite to the movement of the aircraft, but there is also a reason for our approach.

I **think** it is more common to consider the drag force towards the "aircraft" axis (as RocketPy does) when we are working with **rockets**, but this is not a very certain answer. I'm sure @giovaniceotto can give us a good reason for that assumption.

Gui-FernandesBR, 2022-02-13:
The complete resolution for this question can be read on [RocketPy Article](https://ascelibrary.org/doi/abs/10.1061/%28ASCE%29AS.1943-5525.0001331), See Flight and Rocket Model for better compreend it.

_"Aerodynamic forces acting on rockets are usually decomposed into the axial force A and the normal force N"_

closed 2022-02-13 by Gui-FernandesBR