RocketPy-Team/RocketPy #53: Selected Ensemble members

Gui-FernandesBR, 2020-10-18:
Hey, I noticed that the Environment Classe uses, by default, the first Ensemble member available in the Ensemble files. 

But I think that could be intersting if we would use mean valeus based on the ensemble members. For example:
`Env.selectEnsembleMember(Env.ensembleMean)`

This way we could calculate mean and standard deviation from Ensemble members, and I believe the standard deviation of wind speed is actually named wind turbulence (an important input parameter for other rocket simulators).

What do you think about that? Is there a reason for using only one specific ensemble member?

giovaniceotto, 2020-10-19:
Great topic for us to discuss @Gui-FernandesBR. As far as I know, all ensembles have the following characteristics:

- The first ensemble member is a control point, meaning it contains data from the weather simulation using only nominal values without any perturbations.
- All other members are a product of a Monte Carlo analysis, therefore, perturbations are considered.

This is the main reason for the first ensemble member to be chosen by default. However, it may be necessary to check ECMWF and NOAA documentation to verify this.

Using the mean of all ensembles (perhaps without the first member) could be an interesting option, as it may represent the most likely scenario. I have yet to analise how different is the mean of all ensemble members from the first ensemble, but I assume they are close.

The standard deviation of all members can give you some interesting data, however, I do not believe it can be used to estimate turbulence. This would only be possible if the perturbations used to create the ensemble members were due to turbulence alone. In that case, the ensemble variance would be exactly what turbulence is. Unfortunately, I believe the perturbations are derived from uncertainty in measurements. (I have to verify this in ensemble's documentations. ERA5 has a great documentation.)

That being said, let me know what you think about using the mean of all ensemble members for simulations.

I agree it should at least be plotted in `Environment.allInfo()`, along with the standard deviation.

One more thing, if you could link me to documentation from other simulators that use wind turbulence in their models, I would greatly appreciate. I am not sure how this would be used yet.

Gui-FernandesBR, 2024-07-06:
In #605 I made it clearer that the first Ensemble member is the "average" one, and therefore it is the default used when creating an environment with Ensemble models.

[image]

closed 2024-07-12 by Gui-FernandesBR