RocketPy-Team/RocketPy #54: Rocket.addFins() does not support airfoil lift coefficient data with Cl=0

giovaniceotto, 2020-11-08:
**Use case**
Try to add fins to a rocket object using the airfoil option:

```python
FinSet = Europa.addFins(
    n=3, 
    rootChord=0.077, 
    tipChord=47/1000, 
    span=0.104,
    distanceToCM=-0.99586,
    airfoil='NACA0012_Cl.csv'
)
```

Contents of 'NACA0012_Cl.csv' are:

```csv
0, 6.86E-09
0.200712864, 1.05E+00
1.560796326, 0.0
```

**Expected result**
Fins are added properly with no errors  or warnings.

**What happens instead**

```
RuntimeWarning: divide by zero encountered in double_scalars FD = 2 * np.pi * AR / (cn *np.cos(gamac))
RuntimeWarning: invalid value encountered in double_scalars Cnalfa1 = cn * FD * (Af/self.area) * np.cos(gamac) / (2 + FD * ( 1 + (4/FD**2) )**0.5)
```

Furthermore, the resulting `cldata` created has a **nan** _(not a number)_ everywhere a zero is expected.

```python
[[0.00000000e+00 5.18408620e-09]
 [2.00712864e-01 7.19089724e-01]
 [1.56079633e+00            nan]]
```

**Applies to versions:**
Master branch only, no releases yet.

brunosorban, 2021-05-13:
The bug is caused by an error in the implementation of the lift coefficient calculation. This bug distorts the static margin value. A solution is already in progress, but the "airfoil" option should not be used until the  bug is fixed.

giovaniceotto, 2021-05-14:
Great. Let me know if you need anything.

Gui-FernandesBR, 2021-10-21:
@Projeto-Jupiter/back-end

Gui-FernandesBR, 2022-01-31:
@lucasfourier please take a look

Gui-FernandesBR, 2022-02-13:
I believe we can close this one based on #122 awesome implementation,

Many thanks to @MateusStano who did that. Let's rock!

giovaniceotto, 2022-02-13:
I don't believe this is the case.

Gui-FernandesBR, 2022-02-20:
Fixed by @MateusStano on #140 , wait for release v0.10.0

awesome work!!

closed 2022-02-20 by Gui-FernandesBR