RocketPy-Team/RocketPy #55: Rocket.addFins() airfoil argument should support floats and lists

giovaniceotto, 2020-11-08:
**Use case**
Try to add fins to a rocket object using a list as the airfoil argument:

```python
cl_data = [
    [0.00, 0.0],
    [0.20, 1.05],
    [1.00, 1.05],
    [1.57, 0.0],
]

FinSet = TestRocket.addFins(
    n=3, 
    rootChord=0.077, 
    tipChord=47/1000, 
    span=0.104,
    distanceToCM=-0.99586,
    airfoil=cl_data
)
```

**Expected result**
Fins are added properly with no errors  or warnings.
Airfoil argument should support entries in list format, just like other similar RocketPy entries (powerOnDrag, powerOffDrag, thrust, etc.):
- **Lists** should be converted into functions using **Function**.
- **Floats** could be interpreted as a lift coefficient derivative.

**What happens instead**

```shell
Exception has occurred: AttributeError 'list' object has no attribute 'split'
  File "rocketpy\Rocket.py", line 607, in addFins
    read = genfromtxt(airfoil, delimiter = ',')
  File "test.py", line 41, in <module>
    FinSet = TestRocket.addFins(
```

**Applies to versions:**
Master branch only, no releases yet.

Gui-FernandesBR, 2021-10-21:
@Projeto-Jupiter/back-end

Gui-FernandesBR, 2022-02-20:
Fixed bt @MateusStano on #140 . The release v9.10.0 will modify what is needed on the code.

Awesome work!!

closed 2022-02-20 by Gui-FernandesBR