RocketPy-Team/RocketPy #60: Can't understand the meaning of some parameters in the code

aipla999, 2021-03-22:
Beautiful program. But  I have some difficulties when running the  code example in Monte Carlo Dispersion Analysis  section of the reference docs.  I can't understand the meaning of some parameters in the code  so that  I can't simulate my rocket  flight correctly.

 Could you tell me the meaning of    _"m_prop",   "m_aero",    "m_rec",   "lag_rec" ,    "m_se",   "lag_se"_    in  rockst‘s parameters that will be analyzed by Monte Carlo method ? And   how do I determine the second arguments of the Gaussian distribution of the variable? Take this parameter , "m_prop": (1.664, 0.05) ,  for example,  how do I get an appropriate number, such as 0.05？

Besides, from the code, I konw that  mass , a parameter of Rocket class ,  is the total mass of the rocket without propellant, but  why ‘’m_aero‘’ was added twice to get rocket’s mass？

giovaniceotto, 2021-03-22:
It seems like you are write, our [current example notebook](https://rocketpyalpha.readthedocs.io/en/latest/notebooks/dispersion_analysis.html) seems to be outdated and may even contain some bugs.

I will create a new one and update the documentation, trying to fix errors and making it more clear.

giovaniceotto, 2021-03-22:
Let me just advance a couple of things for you.

The analysis parameters are a collection of expected values (and their uncertainties, or standard deviation) that completely defines a rocket flight.
As an assumption, the parameters which define the flight can behave in 3 different ways:
 - the parameter is a completely known and constant value (i.e. number of fins)
 - the parameter can assume certain discrete values with uniform distribution (i.e. the member of an ensemble forecast, which might be any integer from 0 to 9)
 - the parameter is best represented by a normal (gaussian) distribution with a defined expected value and standard deviation

We implement this using a dictionary called `analysis_parameters`, where the key is the name of the parameter and the value is either a tuple or a list, depending on the behavior of the parameter:
 - if the parameter is know, its value is represented as a list with a single entry (i.e. `"number_of_fins: [4]"`)
 - if the parameter can assume certain discrete values with uniform distribution, its values are represented by a list of possible choices (i.e. `"member_of_ensemble_forecast: [0, 1, 2, 3, 4, 5, 6, 7, 8, 9]"`)
 - if the parameter is best represented by a normal (gaussian) distribution, its value is a tuple with the expected value and its standard deviation (i.e. `"rocket_mass": (100, 2)`, where 100 kg is the expected mass, with uncertainty of plus or minus 2 kg)

Here is an example of `analysis_parameters`:

```python
analysis_parameters = {
    # Mass Details
    "rocketMass": (8.257, 0.001), # Rocket's dry mass (kg) and its uncertainty (standard deviation)
    
    # Propulsion Details - run help(SolidMotor) for more information
    "impulse": (1415.15, 35.3),                         # Motor total impulse (N*s)
    "burnOut": (5.274, 1),                              # Motor burn out time (s)
    "nozzleRadius": (21.642/1000, 0.5/1000),            # Motor's nozzle radius (m)
    "throatRadius": (8/1000, 0.5/1000) ,                # Motor's nozzle throat radius (m)
    "grainSeparation": (6/1000, 1/1000),                # Motor's grain separation (axial distance between two grains) (m)
    "grainDensity": (1707, 50),                         # Motor's grain density (kg/m^3)
    "grainOuterRadius": (21.4/1000, 0.375/1000),        # Motor's grain outer radius (m)
    "grainInitialInnerRadius": (9.65/1000, 0.375/1000), # Motor's grain inner radius (m) 
    "grainInitialHeight": (120/1000, 1/1000),           # Motor's grain height (m)

    # Aerodynamic Details - run help(Rocket) for more information
    "inertiaI": (3.675, 0.03675),                       # Rocket's inertia moment perpendicular to its axis (kg*m^2) 
    "inertiaZ": (0.007, 0.00007),                       # Rocket's inertia moment relative to its axis (kg*m^2)
    "radius": (40.45/1000, 0.001),                      # Rocket's radius (kg*m^2)
    "distanceRocketNozzle": (-1.024,0.001),             # Distance between rocket's center of dry mass and nozzle exit plane (m) (negative)
    "distanceRocketPropellant": (-0.571,0.001),         # Distance between rocket's center of dry mass and and center of propellant mass (m) (negative)
    "powerOffDrag": (1, 0.033),                         # Multiplier for rocket's drag curve. Usually has a mean value of 1 and a uncertainty of 5% to 10%
    "powerOnDrag": (1, 0.033),                          # Multiplier for rocket's drag curve. Usually has a mean value of 1 and a uncertainty of 5% to 10%
    "noseLength": (0.274, 0.001),                       # Rocket's nose cone length (m)
    "noseDistanceToCM": (1.134, 0.001),                 # Axial distance between rocket's center of dry mass and nearest point in its nose cone (m)
    "finSpan": (0.077, 0.0005),                         # Fin span (m)
    "finRootChord": (0.058, 0.0005),                    # Fin root chord (m)
    "finTipChord": (0.018, 0.0005),                     # Fin tip chord (m)
    "finDistanceToCM": (-0.906, 0.001),                 # Axial distance between rocket's center of dry mass and nearest point in its fin (m) 

    # Launch and Environment Details - run help(Environment) and help(Flight) for more information
    "inclination": (84.7, 1),                           # Launch rail inclination angle relative to the horizontal plane (degrees)
    "heading": (53, 2),                                 # Launch rail heading relative to north (degrees)
    "railLength": (5.7 , 0.0005),                       # Launch rail length (m)
    "ensembleMember": list(range(10)),                  # Members of the ensemble forecast to be used
    
    # Parachute Details - run help(Rocket) for more information
    "CdSDrogue": (0.349*1.3, 0.07),                     # Drag coefficient times reference area for the drogue chute (m^2)
    "lag_rec": (1 , 0.5),                               # Time delay between parachute ejection signal is detected and parachute is inflated (s)
    
    # Electronic Systems Details - run help(Rocket) for more information
    "lag_se": (0.73, 0.16)                              # Time delay between sensor signal is received and ejection signal is fired (s)
}
```
As you can see, this dictionary is quite different from the current one. Therefore, it will not work with the code you current have. Parts of the notebook will need to be modified to adapt to a particular rocket every time. However, this dictionary is well documented and can help guide you.

As soon as I publish the new version of the example notebook, I will let you know here!

aipla999, 2021-03-24:
Thank you for your explanation. 
It solved the problem that has been bothering me for many days
I hope this software can add the following two functions

  1.  Add a variable output function that output the detail main parameters of rocket design to an excel file.  For example ,  output the array  storing height of rocket  with time in excel file format,  which produced in every timestep,  so that I can analysize the  simulation results more deeply.   I have implemented this part of the function in the code and I will submit it to GitHub after improvement soon .

  2. Calculate  static margin for the whole flight phase of the rocket,  not just before the solidmotor burns out.   
Because the static margin will change with the speed of the rocket.

giovaniceotto, 2021-04-10:
I have enhanced the Monte Carlo dispersion analysis example notebook significantly. You can check it out here: https://rocketpyalpha.readthedocs.io/en/latest/notebooks/dispersion_analysis/dispersion_analysis.html

Let me know if this example contains all you need.

giovaniceotto, 2021-04-10:
@aipla999 Looking forward to your suggested contributions as well. Let me know if you need any help/guidance.

closed 2021-04-10 by giovaniceotto

aipla999, 2021-04-20:
@giovaniceotto   Thank you so much.The new reference documents are so detailed.But it will throw errors when I run the sample code of Monte Carlo Dispersion Analysis.

So,  I made some changes to the code to make it run normally.

First,  at the beginning of the programwe should change "import IPython" to"from IPython.display import display" .  So that we can use the function called display directly.

In the Simulating Each Flight Setting section,we use “display” directly.  The location of this line of code is shown in the figure1.

 

 Second,   in the Simulating Each Flight Setting of Monte Carlo Dispersion Analysis,  we should change the "export_flight_error(flight_setting)" to "export_flight_error(setting)".    The location of this line of code is shown in the figure 2.

Because the export_flight_error function is in the body of the program's for loop， we can infer that it’s parameter should be “setting” instead of”flight_setting” from figure 3.

When I make the above changes to the code, the program can run normally.

aipla999, 2021-04-20:
@giovaniceotto Thank you so much.The new reference documents are so detailed.But it will throw errors when I run the sample code of Monte Carlo Dispersion Analysis.

So,I made some changes to the code to make it run normally.

**First**, at the beginning of the programwe should change "import IPython" to"from IPython.display import display" . So that we can use the function called display directly.

In the Simulating Each Flight Setting&nbsp;section,we use “display” directly.

**Second**, in the Simulating Each Flight Setting of Monte Carlo Dispersion Analysis, we should change the "export_flight_error(flight_setting)" to "export_flight_error(setting)".

Because the export_flight_error&nbsp;function is in the body of the program's for loop， we can infer that it’s parameter&nbsp;should be “setting”&nbsp;instead of”flight_setting” from figure 3.
When I make the above changes to the code, the program can run normally.

giovaniceotto, 2021-04-22:
@aipla999 Thanks for reporting the bugs and already fixing them! I have implemented the fixes and pushed them to master. 

Everything should be working perfectly now. Thanks again!