RocketPy-Team/RocketPy #62: Output the simulation data of every timestep in an excel file in the location you specify

aipla999, 2021-04-20:
@giovaniceotto.  Thank you so much.I have run the Monte Carlo analysis  successfully.And the new reference documents are so comprehensive and detailed.  Besides，I made some changes to the sample code ，so that  I can get simulation data of every timestep and do more in-depth analysis. All data will be saved in an excel file in the location you specify.

The specific methods are as follows:
**First**, make sure the name of instance of your flight calss is TestFlight.   For example,   `TestFlight=Flight(maxTime=7000,rocket=Rocket01,environment=Env,inclination=80,heading=0)`
**Second** ,add the following code  after all your code  or after `TestFlight=Flight(maxTime=7000,rocket=Rocket01,environment=Env,inclination=80,heading=0)`
 
And change the value of filepath_default to where you want to store the data.

Add the following code,and you can save the detailed simulation data in an excel file in the location you specify.
```

# filepath_default='/home/myLinux/Documents/RocketPyOutPut.xlsx'      #This is the only code you need to change.
# wb = openpyxl.Workbook()
# ws = wb.active
# wb.save(filepath_default)
#
# def append_to_excel(header, datalist, filepath = filepath_default) -> None:
#     datalist_list= datalist.tolist()
#     datalist_list.insert(0, header)
#     dataframe = pandas.DataFrame(datalist_list)
#     writer = pandas.ExcelWriter(filepath, mode='w')  
#     data = pandas.read_excel(writer, index_col=None, header=None)
#     data.to_excel(writer, startcol=0, index=None, header=None, sheet_name='sheet1')
#     dataframe.to_excel(writer, startcol=data.shape[1], startrow=1,index=None, header=header, sheet_name='sheet1')
#     writer.save()
#
# paramDict = { 'Time':TestFlight.acceleration.source[:,0],
#                   'x': TestFlight.x.source[:, 1],
#                 'y': TestFlight.y.source[:, 1],
#                 'z or attitude': TestFlight.z.source[:, 1],
#                 'staticMargin': TestFlight.staticMargin.source[:, 1],
#               'Accleration(m/s^2)':TestFlight.acceleration.source[:,1],
# 'Velocit vx':TestFlight.vx.source[:,1],
# 'Velocit vy':TestFlight.vy.source[:,1],
# 'Velocit vz':TestFlight.vz.source[:,1],
# 'Velocit Total':TestFlight.speed.source[:,1],
#
# 'Acceleration x':TestFlight.ax.source[:,1],
# 'Acceleration y':TestFlight.ay.source[:,1],
# 'Acceleration z':TestFlight.az.source[:,1],
#
# 'Euler e0':TestFlight.e0.source[:,1],
# 'Euler e1':TestFlight.e1.source[:,1],
# 'Euler e2':TestFlight.e2.source[:,1],
# 'Euler e3':TestFlight.e3.source[:,1],
#
# 'Euler Precession Angle ψ (°)':TestFlight.psi.source[:,1],
# 'Euler Nutation Angle θ (°)':TestFlight.theta.source[:,1],
# 'Euler Spin Angle φ (°)':TestFlight.phi.source[:,1],
#
# 'Flight Path Angle (°)':TestFlight.pathAngle.source[:,1],
# 'Rocket Attitude Angle (°)':TestFlight.attitudeAngle.source[:,1],
# 'Lateral Attitude Angle (°)':TestFlight.lateralAttitudeAngle.source[:,1],
#
# 'Angular Velocity -w1(rad/s)':TestFlight.w1.source[:,1],
# 'Angular Velocity -w2(rad/s)':TestFlight.w2.source[:,1],
# 'Angular Velocity -w3(rad/s)':TestFlight.w3.source[:,1],
# 'Angular Acceleration -alpha1(rad/s²)':TestFlight.alpha1.source[:,1],
# 'Angular Acceleration -alpha2(rad/s²)':TestFlight.alpha2.source[:,1],
# 'Angular Acceleration -alpha3(rad/s²)':TestFlight.alpha3.source[:,1],
#
# 'Rail Buttons Normal Force  Upper Rail Button(N)':TestFlight.railButton1NormalForce.source[:,1],
# 'Rail Buttons Normal Force  Lower Rail Button(N)':TestFlight.railButton2NormalForce.source[:,1],
# 'Rail Buttons Shear Force  Upper Rail Button(N)':TestFlight.railButton1ShearForce.source[:,1],
# 'Rail Buttons Shear Force  Lower Rail Button(N)':TestFlight.railButton2ShearForce.source[:,1],
# 'Aerodynamic Lift Resultant Force  Resultant':TestFlight.aerodynamicLift.source[:,1],
# 'Aerodynamic Lift Resultant Force  R1':TestFlight.R1.source[:,1],
# 'Aerodynamic Lift Resultant Force  R2':TestFlight.R2.source[:,1],
# 'R3':TestFlight.R3.source[:,1],
#
# 'Aerodynamic Drag Force (N) ':TestFlight.aerodynamicDrag.source[:,1],
# 'Aerodynamic Bending Resultant Moment Resultant ':TestFlight.aerodynamicBendingMoment.source[:,1],
# 'Aerodynamic Bending Resultant Moment M1':TestFlight.M1.source[:,1],
# 'M2':TestFlight.M2.source[:,1],
# 'M3':TestFlight.M3.source[:,1],
# 'Aerodynamic Spin Moment  (N m)':TestFlight.aerodynamicSpinMoment.source[:,1],
#
# 'Kinetic Energy  Kinetic Energy Components（J）':TestFlight.kineticEnergy.source[:,1],
# 'rotationalEnergy':TestFlight.rotationalEnergy.source[:,1],
# 'Translational Energy':TestFlight.translationalEnergy.source[:,1],
# 'total Energy':TestFlight.totalEnergy.source[:,1],
# 'potential Energy':TestFlight.potentialEnergy.source[:,1],
# 'Thrust Absolute Power':TestFlight.thrustPower.source[:,1],
# 'Drag Absolute Power（value <0）':-TestFlight.dragPower.source[:,1],
#
# 'Mach Number':TestFlight.MachNumber.source[:,1],
# 'Reynolds Number':TestFlight.ReynoldsNumber.source[:,1],
# 'dynamic Pressure':TestFlight.dynamicPressure.source[:,1],
# 'total Pressure':TestFlight.totalPressure.source[:,1],
# 'Static Pressure':TestFlight.pressure.source[:,1],
# 'Angle of Attack':TestFlight.angleOfAttack.source[:,1],

# 'Pressure at Rocket\'s Altitude    Altitude(m)':TestFlight.z.source[:,1],
# 'Pressure at Rocket\'s Altitude    Pressure(Pa':TestFlight.pressure.source[:,1],
#
# 'static Margin':TestFlight.staticMargin.source[:,1]}
#
# for key in paramDict.keys():
#     print(key)
#     append_to_excel(key, paramDict[key])
# # print('The data has been successfully written into an excel file.')
```

giovaniceotto, 2021-04-22:
This feature can be incredibly useful! Very nice implementation. I will try it here and let you know how it goes.

Gui-FernandesBR, 2021-11-10:
> This feature can be incredibly useful! Very nice implementation. I will try it here and let you know how it goes.

@giovaniceotto did you tried?

Gui-FernandesBR, 2022-03-23:
Solved by #146 , many thanks to Giovani in this case, and of course thanks @aipla999 as well

closed 2022-03-23 by Gui-FernandesBR