RocketPy-Team/RocketPy #82: An error occurred while simulating the flight of the two-stage rocket

aipla999, 2021-08-12:
**Describe the bug**
Recently ，I found that there are bugs when using RocketPy to simulate a two-stage rocket .
OS: Windows      
RocketPy  Version: 0.97

**Problem**
The following code only shows that the initial speed of the second stage rocket in X direction and Y direction must be set to 0,
otherwise the simulation results of the second stage rocket will not be obtained due to too long simulation time

**Simulation condition assumptions**
We assume that the shape and motor of the first stage rocket and the second stage rocket are the same,
but the initial state of launch is different. The first stage rocket is launched from the ground at a standstill,
and the second stage rocket is fired at the highest point of the first stage rocket

**Here is the complete demo code.**
`
import datetime
from rocketpy import Environment
from rocketpy import Flight
from rocketpy import Rocket
from rocketpy import SolidMotor

Env = Environment(
    railLength=5.2,
    latitude=32.990254,
    longitude=-106.974998,
    elevation=1400,
    date=(2020, 3, 4, 12) # Tomorrow's date in year, month, day, hour UTC format
)
tomorrow = datetime.date.today() + datetime.timedelta(days=1)
Env.setDate((tomorrow.year, tomorrow.month, tomorrow.day, 6))
Env.setAtmosphericModel(type='StandardAtmosphere')

Pro75M1670 = SolidMotor(
    thrustSource="E:\\rocketpy\\mydeepin\\Desktop\\rockstpy096\\rocketpyDocument\\data\motors\\Cesaroni_M1670.eng",
    burnOut=3.9,
    grainNumber=5,
    grainSeparation=5/1000,
    grainDensity=1815,
    grainOuterRadius=33/1000,
    grainInitialInnerRadius=15/1000,
    grainInitialHeight=120/1000,
    nozzleRadius=33/1000,
    throatRadius=11/1000,
    interpolationMethod='linear'
)

Calisto = Rocket(
    motor=Pro75M1670,
    radius=127/2000,
    mass=19.197-2.956,
    inertiaI=6.60,
    inertiaZ=0.0351,
    distanceRocketNozzle=-1.255,
    distanceRocketPropellant=-0.85704,
    powerOffDrag='E:\\rocketpy\\mydeepin\\Desktop\\rockstpy096\\rocketpyDocument\\data\calisto\\powerOffDragCurve.csv',
    powerOnDrag='E:\\rocketpy\\mydeepin\\Desktop\\rockstpy096\\rocketpyDocument\\data\calisto\\powerOnDragCurve.csv'
)

Calisto.setRailButtons([0.2, -0.5])
NoseCone = Calisto.addNose(length=0.55829, kind="vonKarman", distanceToCM=0.71971)
FinSet = Calisto.addFins(4, span=0.100, rootChord=0.120, tipChord=0.040, distanceToCM=-1.04956)
Tail = Calisto.addTail(topRadius=0.0635, bottomRadius=0.0435, length=0.060, distanceToCM=-1.194656)

TestFlight1 = Flight(rocket=Calisto, environment=Env, inclination=85, heading=0,terminateOnApogee=True)
TestFlight1.info()
print("###################### run 2ndStage simulation ############################")
# The 1stStage will terminate On Apogee,and its Apogee State of  will be the initial state of 2ndStage
initSolution2Stage = [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0.0]
# initSolution2Stage[0]=TestFlight1.solution[-1][1]   # Init_time  of 2ndStage
initSolution2Stage[1] = TestFlight1.solution[-1][1]   # Init_x  of 2ndStage
initSolution2Stage[2] = TestFlight1.solution[-1][2]   # Init_y  of 2ndStage
initSolution2Stage[3] = TestFlight1.solution[-1][3]   # Init_z  of 2ndStage

# Only if Init_vx=0 and Init_vy=0,running the program can get the result in a few seconds
# Otherwise, it will take too long to get results
initSolution2Stage[4] = TestFlight1.solution[-1][4]   # Init_vx of 2ndStage
initSolution2Stage[5] = TestFlight1.solution[-1][5]   # Init_vy of 2ndStage

initSolution2Stage[6] = TestFlight1.solution[-1][6]   # Init_vz of 2ndStage
TestFlight2 = Flight(rocket=Calisto, environment=Env, inclination=85, heading=0,initialSolution=initSolution2Stage)
TestFlight2.info()
`

lucasfourier, 2021-08-12:
How are you doing @aipla999? 

The development team is looking into it and we will do our best to fix this issue as soon as possible! We will come back to report updates and/or ask for further details. Thanks for pointing this issue.

aipla999, 2021-08-17:
When the velocity in X or Y direction is not zero in the initial launch state of the  rocket in the second episode, the problem of too small time step will appear in the solution of ODE with the progress of simulation.

If I set the minimum time step of the simulation to 0.01 second,   (For example ,`**TestFlight2 = Flight(maxTime=7000, ******, minTimeStep=0.01   )**`),   the following errors will occur during simulation
_**D:\anacondaInstall\lib\site-packages\scipy\integrate\_ode.py:1350: UserWarning: lsoda: Repeated convergence failures (perhaps bad Jacobian or tolerances) warnings.warn('{:s}: {:s}'.format(self.__class__.__name__,**_
  
However, the whole simulation can still continue, but the coordinates of the rocket's impact point are obviously incorrect.

Lucas-KB, 2021-08-17:
# The problem with the original code

The main problem we identified with your original code was regarding the initial orientation of the rocket given in the initial solution. It is not valid, i.e. it does not represent any orientation since $e_0^2 +e_1^2 + e_2^2 + e_3^2 \neq 1$. A valid option would be (1, 0, 0, 0), or even better take the last orientation of the first stage: `initSolution2Stage[7:11] = TestFlight1.solution[1][7:11]`. We intend to add a warning to avoid similar situations.

# The problems found in RocketPy

While investigating this particular problem, we found three issues:

  - The simulation does not work when the initial time differs from zero;
  - Some of the energy plots of the second stage may not show up unexpectedly. They can be plotted separately, for example to plot the Kinetic Energy: `TestFlight2.kineticEnergy.plot()`;
  - The reported apogee for the second stage will likely be incorrect ($t=0.00s$). This false reporting is likely due to a small drop in altitude at the beggining of the seconde stage, since the thrust from SolidMotor usually starts at low values (which can't overcome the forces resultant from gravity). The simulation, however, works as expected and if you desire to know the apogee, you could run: `TestFlight2.z.source.max() - Env.elevation`.

We plan on fixing all these issues soon, but for now we offer a temporary solution below.

# A viable alternative

```Python
import datetime
from rocketpy.Function import Function
from rocketpy import Environment
from rocketpy import Flight
from rocketpy import Rocket
from rocketpy import SolidMotor

# Set the environment
Env = Environment(
    railLength=5.2,
    latitude=32.990254,
    longitude=-106.974998,
    elevation=1400,
    date=(2020, 3, 4, 12),  # Tomorrow's date in year, month, day, hour UTC format
)
tomorrow = datetime.date.today() + datetime.timedelta(days=1)
Env.setDate((tomorrow.year, tomorrow.month, tomorrow.day, 6))
Env.setAtmosphericModel(type="StandardAtmosphere")

# Set the motor
Pro75M1670 = SolidMotor(
    thrustSource="..\RocketPy\data\motors\Cesaroni_M1670.eng",
    burnOut=3.9,
    grainNumber=5,
    grainSeparation=5 / 1000,
    grainDensity=1815,
    grainOuterRadius=33 / 1000,
    grainInitialInnerRadius=15 / 1000,
    grainInitialHeight=120 / 1000,
    nozzleRadius=33 / 1000,
    throatRadius=11 / 1000,
    interpolationMethod="linear",
)

# Set the rocket
Calisto = Rocket(
    motor=Pro75M1670,
    radius=127 / 2000,
    mass=19.197 - 2.956,
    inertiaI=6.60,
    inertiaZ=0.0351,
    distanceRocketNozzle=-1.255,
    distanceRocketPropellant=-0.85704,
    powerOffDrag="..\RocketPy\data\calisto\powerOffDragCurve.csv",
    powerOnDrag="..\RocketPy\data\calisto\powerOnDragCurve.csv",
)

Calisto.setRailButtons([0.2, -0.5])
NoseCone = Calisto.addNose(length=0.55829, kind="vonKarman", distanceToCM=0.71971)
FinSet = Calisto.addFins(
    4, span=0.100, rootChord=0.120, tipChord=0.040, distanceToCM=-1.04956
)
Tail = Calisto.addTail(
    topRadius=0.0635, bottomRadius=0.0435, length=0.060, distanceToCM=-1.194656
)

# Set the first stage
TestFlight1 = Flight(
    rocket=Calisto, environment=Env, inclination=85, heading=0, terminateOnApogee=True
)
TestFlight1.info()
print("###################### run 2nd Stage simulation ############################")

# The 1st Stage will terminate On Apogee,and its Apogee State of will be the initial state of 2ndStage

TestFlight2 = Flight(
    rocket=Calisto,
    environment=Env,
    inclination=85,
    heading=0,
    initialSolution=[0] + TestFlight1.solution[-1][1:],
    verbose=True,
)
TestFlight2.info()
```

aipla999, 2021-08-18:
Thank you very much. I have modified my code and got the desired simulation data.

lucasfourier, 2021-08-18:
@aipla999 We are glad that we were able to help! Also, we thank you very much for helping us identify some important issues regarding the code.

closed 2021-08-18 by lucasfourier