RocketPy-Team/RocketPy #111: A new place for utility functions

giovaniceotto, 2021-11-10:
@Gui-FernandesBR mentioned some candidates to be included in the utilities.py module being introduced in #108:

> Environment.py:
> `decimalDegressToArcSeconds(self, angle)`
> 
> Flight.py:
> `calculateFinFlutterAnalysis(self, finThickness, shearModulus)`
> `exportPressures(self, fileName, timeStep)`

These could help give more "body" to this new module.

Also, @Gui-FernandesBR recommends creating a simple example usage for these functions. At least for me, it is not easy to visualize how I could use the new capabilities.

Lucas-KB, 2021-11-10:
I do agree with most of what @Gui-FernandesBR and @giovaniceotto suggested, but I am not sure if `exportPressures` fits here. I believe that one belongs in the `Flight` class, just like `exportEng` belongs in the `SolidMotor` class, does it make sense to you?

I suspect you might have thought of this because `exportPressures` feels a little weird in the sense that it is special for pressures, because it was created to help the electronics team at Projeto Jupiter. What may be a way around this is to implement a more generic function that exports any number of selected simulation results to a csv file (with all that time step thing that exists in `exportPressures`). What do you think?

giovaniceotto, 2021-11-10:
@Lucas-KB, I completely agree with you.

Gui-FernandesBR, 2021-11-10:
@Lucas-KB awesome comment. I think we should go for that, a generic function to export simulations data to .csv files may be very useful. It seems @aipla999 was creating similar things at #62 

Just adding an idea: We could have (1) a standard set of data to be exported and also (2) the possibility of customizing what you want to export. What do you think?

Lucas-KB, 2021-11-10:
@Gui-FernandesBR I did not know about #62, seems very interesting. Do you have any suggestion on what could be in the standard set of data? I believe the obvious answer would be to output everything, any different ideas?

Gui-FernandesBR, 2021-11-10:
Weel I think the _u vector_ would be enough (time, x, y, z, vx, vy, vz, ax, ay, az, etc.)

Just for comparision reasons, OpenRocket currently can export all the following data when requested:

> Time (s);  Altitude (m);  Vertical velocity (m/s);  Vertical acceleration (m/s²);  Total velocity (m/s);  Total acceleration (m/s²); 
 Position East of launch (m);  Position North of launch (m);  Lateral distance (m);  Lateral direction (°);  Lateral velocity (m/s); 
 Lateral acceleration (m/s²);  Latitude (°);  Longitude (°);  Gravitational acceleration (m/s²);  Angle of attack (°);  Roll rate (°/s); 
 Pitch rate (°/s);  Yaw rate (°/s);  Mass (g);  Propellant mass (g);  Longitudinal moment of inertia (kg·m²);  Rotational moment of inertia (kg·m²);  CP location (mm);  CG location (mm);  Stability margin calibers (?);  Mach number (?);  Reynolds number (?);  Thrust (N);  Drag force (N);  Drag coefficient (?);  Axial drag coefficient (?);  Friction drag coefficient (?);  Pressure drag coefficient (?);  Base drag coefficient (?);  Normal force coefficient (?);  Pitch moment coefficient (?);  Yaw moment coefficient (?);  Side force coefficient (?);  Roll moment coefficient (?);  Roll forcing coefficient (?);  Roll damping coefficient (?);  Pitch damping coefficient (?);  Coriolis acceleration (m/s²); Reference length (mm);  Reference area (m²);  Vertical orientation (zenith) (°);  Lateral orientation (azimuth) (°);  Wind velocity (m/s); Air temperature (°C);  Air pressure (mbar);  Speed of sound (m/s);  Simulation time step (s);  Computation time (s)

I don't think we need all these data at once, starting with simple one would be fine for me.
But of course I open to suggestions as well.

Gui-FernandesBR, 2022-02-01:
Is that possible that the following functions be implemented on utility functions?

* CompareRockets
* ApogeeByMass
* OutOfRailByMass

lucasfourier, 2022-02-01:
In my opinion, "CompareRockets" is the perfect candidate and would suit just fine in utilities.py. Regarding the other two, I am not so sure.

closed 2023-07-09 by Gui-FernandesBR