RocketPy-Team/RocketPy #114: Getting Started Notebook Fails

giovaniceotto, 2021-12-08:
**Describe the bug**

When running the getting started notebook ([which can be done via Colab](https://colab.research.google.com/github/giovaniceotto/rocketpy/blob/master/docs/notebooks/getting_started_colab.ipynb)), it fails and throws and error in the Dynamic Stability Analysis section.

The error happens because a typo was fixed in every file of the master branch, including in the notebook. The typo in question was the incorrect spelling "Costum" instead of "Custom", which affects the "CustomAtmosphere" argument that can be used in the `Environment.setAtmosphericModel` method. However, the current version of rocketpy released on [PYPI (0.9.8)](https://pypi.org/project/rocketpy/) does not yet incorporate this typo fix.

**To Reproduce**

1. Access the Getting Started notebook via [Google Colab](https://colab.research.google.com/github/giovaniceotto/rocketpy/blob/master/docs/notebooks/getting_started_colab.ipynb)
2. Run the entire notebook (all the cells, in order)
3. Notice `ValueError: Unknown model type` after running the Dynamic Stability Analysis cell

**Expected behavior**

The entire notebook should run without errors.

**Screenshots**

[image]

**Solution Sugestion**

Release a new version!

Gui-FernandesBR, 2021-12-08:
Tks for sharing this!

Version Release will be adressed as soon as possible, I'll handle it myself.

Gui-FernandesBR, 2022-01-04:
Hey, sorry for the failure on the asap part.
Still, I'd like to handle it myself anyway, I need to learn how to use the automatic release on PyPI package.
I'm creating here a deadline of 5days, so until 09th january this may be released.

@giovaniceotto can you help me to understand if we need to merge anything from "develop" do "master" before realeasing the new version? If positive, let's select the pull request and create the merge.

giovaniceotto, 2022-01-05:
@Gui-FernandesBR, I do believe we may have a couple of things in the develop branch that could be merged into master before the new version. Could we discuss it further over discord, together with the automatic release?

Gui-FernandesBR, 2022-01-06:
Last release was created on 16th August. Since that day, the following PR were merged on develop branch: 
- #116 - Roll moment Calculation added
- #108 - Simple typos
- #105 - Fixed .eng file bug;
I'm proposing pulling these PRs into marge branch before launching the release on 8th January. The first one gives me a litte question: Should we wait before more users test the functino or should we go and already implement it on master version of RocketPy? The 2nd and 3rd are really simple so I think it is easy to make the decision.

@MrGribel could you please check and validate the proposal giving your OK (or NOK) for each line?
We need this before proceeding with the process of merging develop into master

MrGribel, 2022-01-07:
Master branch should contain stable code. I doubt we have many users extensively testing our dev branch since it is not the default one, meaning we ourselves need to check if it is ready to deploy. Everything seems to be running well on my end and imo these PRs are too unrelated to have broken anything hidden due to conflict. 

Just a thing: develop currently has a CI build error that should probably be adressed first. After that, feel free to merge and create a new release.

Gui-FernandesBR, 2022-01-09:
Works fine now, tks to all team!

closed 2022-01-09 by Gui-FernandesBR