RocketPy-Team/RocketPy #115: Stable rockets with negative stability margins?

giovaniceotto, 2021-12-08:
**Describe the bug**

Rockets with slightly negative stability margins seem stable in RocketPy

**To Reproduce**

Run the following code:

```python
# Helper class
from rocketpy import Environment, SolidMotor, Rocket, Flight, Function

# Prepare Motor
Pro75M1670 = SolidMotor(
    thrustSource="../../data/motors/Cesaroni_M1670.eng",
    burnOut=3.9,
    grainNumber=5,
    grainSeparation=5/1000,
    grainDensity=1815,
    grainOuterRadius=33/1000,
    grainInitialInnerRadius=15/1000,
    grainInitialHeight=120/1000,
    nozzleRadius=33/1000,
    throatRadius=11/1000,
    interpolationMethod='linear'
)

# Prepare Rocket
Calisto = Rocket(
    motor=Pro75M1670,
    radius=127/2000,
    mass=19.197-2.956,
    inertiaI=6.60,
    inertiaZ=0.0351,
    distanceRocketNozzle=-1.255,
    distanceRocketPropellant=-0.85704,
    powerOffDrag='../../data/calisto/powerOffDragCurve.csv',
    powerOnDrag='../../data/calisto/powerOnDragCurve.csv'
)
Calisto.setRailButtons([0.2, -0.5])

Nose = Calisto.addNose(length=0.55829, kind="vonKarman", distanceToCM=0.71971)
FinSet = Calisto.addFins(4, span=0.1, rootChord=0.120, tipChord=0.040, distanceToCM=-1.04956)
Tail = Calisto.addTail(topRadius=0.0635, bottomRadius=0.0435, length=0.060, distanceToCM=-1.194656)

# Prepare Environment
Env = Environment(5.2, 9.8)
Env.setAtmosphericModel(type='CostumAtmosphere', wind_v=-10)

# Simulate Different Static Margins by Varying Fin Position
simulation_results = []

for factor in [0.1, 0.3, 0.5, 0.8, 1.0]:
    # Modify rocket fin set by removing previous one and adding new one
    Calisto.aerodynamicSurfaces.remove(FinSet)
    FinSet = Calisto.addFins(4, span=0.1, rootChord=0.120, tipChord=0.040, distanceToCM=-1.04956*factor)
    # Simulate
    TestFlight = Flight(rocket=Calisto, environment=Env, inclination=90, heading=0, maxTimeStep=0.01, maxTime=2.5, terminateOnApogee=True)
    # Post process flight data
    TestFlight.postProcess()
    # Store Results
    staticMarginAtIginition = Calisto.staticMargin(0)
    staticMarginAtOutOfRail = Calisto.staticMargin(TestFlight.outOfRailTime)
    staticMarginAtSteadyState = Calisto.staticMargin(TestFlight.tFinal)
    simulation_results += [(TestFlight.attitudeAngle, 'Static Margin of {:1.2f} c'.format(staticMarginAtOutOfRail))]

Function.comparePlots(simulation_results, lower=0, upper=2.5, xlabel='Time (s)', ylabel='Attitude Angle (deg)')
```

**Expected behavior**

Negative stability margins should be unstable.... right?

**Screenshots**
Here is a plot of the angle that the Rocket makes with the horizontal plane after leaving the rail. Notice how the green and orange trajectories are stable even thought their stability margin is negative, 

[image]

lucasfourier, 2022-01-29:
I hope that this does not continue to happen for much longer. The back end team is going to look into it very soon.

Gui-FernandesBR, 2022-02-13:
Hey, when testing the current version available on develop branch, I unfortunately got the same bug happening again...

Could one of you two check if the error is from my part or actually #125 didn't resolved this bug? @Lucas-KB @giovaniceotto 

[image]

Lucas-KB, 2022-02-13:
Actually, @Gui-FernandesBR, the bug was reintroduced when you merged pull request #122 changes back into the [bug/issue-115-static-margin-stability](https://github.com/Projeto-Jupiter/RocketPy/tree/bug/issue-115-static-margin-stability) branch (commit 3dd62c8a6962f2f6c8e2ffe4a112a4818e54e2d1). To fix it, we need to remove the `abs` from line 1327 of `Flight.py`.

Lucas-KB, 2022-02-13:
Actually, there might be a problem even fixing that. I am checking that now, will come back with more information in a bit.

giovaniceotto, 2022-02-13:
Solved by PR #125:

[image]

closed 2022-02-13 by giovaniceotto