RocketPy-Team/RocketPy #124: Convert To-do list into Github Projects

Gui-FernandesBR, 2022-01-31:


giovaniceotto, 2022-01-31:
Feel free to delete/archive old projects which haven't being used in a while.

Gui-FernandesBR, 2022-02-01:
I just implemented all important items from TODO.md into project "RocketPy action log",
In the near future we may remove TODO.md file from our repository in order to have only a single source of truth

Gui-FernandesBR, 2022-02-01:
Also, now our project also has issues and Pull Rquests from the famous Projeto-Jupiter/Hackathon repository, hope this will help us to improve RocketPy funcionalities faster

giovaniceotto, 2022-02-12:
@Gui-FernandesBR, can we close this?

Gui-FernandesBR, 2022-02-13:
Sure thing!

Github project give us a way better view than the old TODO.md file. Let's rock!!

[image]

closed 2022-02-13 by Gui-FernandesBR