RocketPy-Team/RocketPy #126: Getting started running your first simulation doesn't tell that netCDF4 must be installed

skrobchik, 2022-02-09:
netCDF4 is not automatically installed.

If user blindly follows the getting started tutorial, they will stumble upon an error when running 
```
Env = Environment(
    railLength=5.2,
    latitude=32.990254,
    longitude=-106.974998,
    elevation=1400,
    date=(2020, 3, 4, 12) # Tomorrow's date in year, month, day, hour UTC format
) 

Env.setAtmosphericModel(type='Forecast', file='GFS')
```

which results in the following error:

```
[NameError: name 'netCDF4' is not defined]()
```

There should be a reminder in the tutorial telling the user to install the optional package

giovaniceotto, 2022-02-09:
Thanks for bringing this to our attention @skrobchik. This was brought to our attention recently and as you mentioned, the main problem is that netCDF4 is an optional package.

We brainstormed several solutions:

- Make the note on our [docs](https://docs.rocketpy.org/en/latest/user/requirements.html#required-packages) regarding netCDF4 more visible, by putting it in a big red box. Also add it to the installation page, besides the requirements page where it sits now. This is quite easy to do.
- Instead of raising a bad error, clearly advise users without netCDF4 how to install it when rocketpy needs it and the user does not have it.
- Add netCDF4 as a requirement. This makes sure that it is installed when rocketpy is installed. Should solve any hiccups for most users, however, in a few cases, installing RocketPy will not be possible anymore without changing the source code.
- Get rid of netCDF4 and move to xarray (perhaps even herbie-data). This can be a great option but requires significant work.

And you just mentioned a fifth option, which is to also remind users to install it in our example. Anyways, I agree that having a package be an optional requirement while it is used in the first example that the user meets is not quite a good experience.

@Gui-FernandesBR, shall we make this issue a priority to solve it quickly?

skrobchik, 2022-02-09:
I think checking for availability at runtime and instructing the user to install the package instead of throwing a bad error would be a good first solution. At least for the way I encountered it. Something like:
```
This feature of rocketpy requieres "netCDF4" which was not found. You can install it with: pip install netCDF4
```

Gui-FernandesBR, 2022-02-10:
Tks for openig this issue @skrobchik . I believe raising a warning as you suggest is better option here, would you like to create the pull request implementing that? If you don't feel confortable doing that, I can ask someone from our team to help you.
Honestly would be an honor for us if you try it, be welcome

Again tks for sharing it with us.

@giovaniceotto do you think it's still importnat adding a comment on "getting started"  to remind using netCDF4 before using environment models?

skrobchik, 2022-02-10:
#127

giovaniceotto, 2022-02-10:
> Tks for openig this issue @skrobchik . I believe raising a warning as you suggest is better option here, would you like to create the pull request implementing that? If you don't feel confortable [...]

With the new implementation, I don't think it would be necessary.

closed 2022-02-11 by giovaniceotto