RocketPy-Team/RocketPy #152: Keep it Simple, Stupid: Simpler Parachute Triggers

giovaniceotto, 2022-03-06:
## Is your feature request related to a problem? Please describe.

RocketPy supports parachute trigger functions, which can be quite nice to test ejection activation algorithms. **However, when running simple simulations with parachutes, having to code a trigger function can be an unnecessary overhead.** 

## Describe the solution you'd like

I would like the `trigger` argument of the `Flight.addParachute` method to support, besides a callable, two other types of input:

- a string `'apogee'`, in which case, the parachute would be ejected at apogee (intuitive I believe)
- a numeric value, such as `1400`, in which case the parachute would be ejected at an altitude (above sea level, ASL) corresponding to the numeric value in meters.

Here is a code snippet showcasing both:

```python
Main = Calisto.addParachute(
    "Main",
    CdS=10.0,
    trigger=800,
    samplingRate=105,
    lag=1.5,
    noise=(0, 8.3, 0.5),
)

Drogue = Calisto.addParachute(
    "Drogue",
    CdS=1.0,
    trigger='apogee',
    samplingRate=105,
    lag=1.5,
    noise=(0, 8.3, 0.5),
)
```

## Implementation details

Internally, the `Rocket.addParachute` method may convert both types of new inputs into trigger functions (callables) which will continue to work with the Flight Class without the need of modifying it.

## Additional context

Such a feature would facilitate the creation of parachutes, making RocketPy easier to use. This is particularly important for inexperienced users. Furthermore, the advanced feature of using a trigger function would keep working exactly as it does today. 

It is also important to note that this idea originated when using RocketPy in MATLAB. Currently, a function needs to be defined in Python inside MATLAB to create the triggers, which is awfully bad for user experience, since it can only be done with lambdas. Therefore, such a feature would also greatly enhance the experience for MATLAB users.

Gui-FernandesBR, 2023-05-25:
Not finished yet

MateusStano, 2023-05-25:
All done in #345

closed 2023-05-25 by MateusStano