RocketPy-Team/RocketPy #154: Integrating control measure on the airframe

0x6f736f646f, 2022-03-09:
**Is your feature request related to a problem? Please describe.**

If we have a reaction wheel or a TVC is there a way we can be able to determine the flight profile while incorporating our control methods?

**Additional context**

Being able to integrate control methods

giovaniceotto, 2022-03-11:
Thanks for the feature request @0x6f736f646f! RocketPy can support some types of controllers currently, but TVC (Thrust Vector Control) and reaction wheels are not currently supported.

We can start working on this feature request and create a new branch for it. Implementation shouldn't be hard, and a couple of weeks should be enough to finish it. However, we would need some help to validate our code, especially with people with more expertise in control laws.

Would you be willing to help on validation and/or implementation?

0x6f736f646f, 2022-03-11:
The whole team @nakujaproject is willing to help.

giovaniceotto, 2022-03-11:
I believe you are in our discord server, is that right?

I'll reach out over there to continue our conversation!

0x6f736f646f, 2022-03-11:
Yes. No problem

Gui-FernandesBR, 2022-03-21:
Adding @MateusStano to this thread since he is our expert on Controlling rockets with more than just regular fins !!

giovaniceotto, 2022-03-22:
@MateusStano, do you mind sending me your current implementation of control laws?

Gui-FernandesBR, 2022-08-23:
found a good resource that could help us: https://github.com/Jmack66/TVC-Sim