RocketPy-Team/RocketPy #155: Export to KML

0x6f736f646f, 2022-03-16:
**Is your feature request related to a problem? Please describe.**
Export the flight trajectory to KML format so that it can be viewed on google earth

**Describe the solution you'd like**
Export flight path to a format readable by google earth

**Additional context**
None

Gui-FernandesBR, 2022-03-16:
Hi @0x6f736f646f ! Glad you rose this issue, we currently have some code lines that export Monte Carlo simulation results to a kml file. I believe we could adjust the code to export the flight path and then we implement on our master branch.

I'm currently responsible for that and hope to work on this assingment during next weeks (as you can see at https://github.com/Projeto-Jupiter/Hackathon/pull/93/files ).

Let me know if you want to join us and try some code lines by yourself, that could be your first contribution to our repository and, honestly, I think it would be awesome having you on contributors list!! :)

giovaniceotto, 2022-03-16:
Seems like a good feature to integrate the `Flight.exportData` method. 

To help with our future implementation, I'll already lay out some guidelines:

1. Initialize three new post process attributes in the Flight class: `Flight.elevation`, `Flight.latitude` and `Flight.longitude` in the `Flight.__init_post_process_variables` method. They are all instances of the class `Function`, and are a function of time.

2. Compute the values (as a function of time) of these new three attributes in the `Flight.post_process` method.

3. Create a new method `Flight.exportKML` which uses the attributes created above to format a KML file.

4. The user also has the option to export elevation, latitude and longitude to a csv file using `Flight.exportData`. No code modifications in this method is needed for this to work.

This Stack overflow question can help with the calculation needed to compute latitude and longitude: https://stackoverflow.com/questions/7222382/get-lat-long-given-current-point-distance-and-bearing

Gui-FernandesBR, 2022-03-22:
Ok then, the above code is good enough and you solve our problem here. It's actually using harvisine formula to find geodesic coordinates of a point distanced from one other point that we know the coordinates already. Harvisine uses an assumption that the earth is an sphere, which is ok for preliminary analisis. Perhaps in the future we can use Vincents formulation, which can take into account any ellipsiod as the shape of earth (really cool!)

In the next weeks I'll be pulling some code lines. And I need help from @Projeto-Jupiter/rocketpy-team on our next working session (tomorrow at 22hrs UTC)

Gui-FernandesBR, 2022-06-17:
#186 solved this issue, @0x6f736f646f thanks for raising such important feature request! 

We will release a new version on next weeks/month, but it can already be used on develop branch

closed 2022-06-17 by Gui-FernandesBR