RocketPy-Team/RocketPy #156: Acceleration data to trigger parachutes

giovaniceotto, 2022-03-16:
**Is your feature request related to a problem? Please describe.**

Parachute trigger functions only have access to pressure and the state vector, without acceleration data.

**Describe the solution you'd like**

Acceleration data, with noise.

Gui-FernandesBR, 2022-03-21:
@FranzYuri can you please take a look at this since it's releated to parachute functions? I think you'll like it!! 

Might not be too complicated, but let's use some minutes during the week to discuss if needed

giovaniceotto, 2022-08-17:
@MateusStano, these may help you:

- https://github.com/scipy/scipy/commit/7e2ddf86c5f8df9642f330e4a1d09f838e953306
- https://computing.llnl.gov/sites/default/files/ODEPACK_pub2_u113855.pdf (page 93)

ViniciusCMB, 2025-11-28:
Hi @giovaniceotto ! I'm interested in working on this issue for the recruitment process.

Since I work with embedded systems in my rocketry team, adding acceleration-based triggers feels very relevant to me (especially because we actually faced issues with our parachute triggering during the last LASC).

I am currently analyzing `flight.py` to understand the best way to access the acceleration data (from `u_dot`) and pass it to the parachute triggers.

Let me know if I can proceed with this!

Gui-FernandesBR, 2025-11-28:
> Hi @giovaniceotto ! I'm interested in working on this issue for the recruitment process.
>
> Since I work with embedded systems in my rocketry team, adding acceleration-based triggers feels very [...]

Hello @ViniciusCMB I will try to guide you with these.

Since this issue was raised, a lot has already changed in this lib. It's possible that we no longer need to solve this one. I need to read it more carefully through the weekend and then get back to you asap.

ViniciusCMB, 2025-11-29:
Thanks for the heads-up!

I understand, a lot changes in 3 years. I will wait for your feedback after the weekend then.

In the meantime, I might take a look at other "good first issues" just to familiarize myself with the codebase and maybe tackle something else if this one turns out to be deprecated.

Looking forward to hearing from you.

ViniciusCMB, 2025-11-29:
Thank you for the guidance!

I appreciate the heads-up regarding the complexity of #156. Following your recommendation, I'll tackle issue #242

Gui-FernandesBR, 2025-11-29:
@ViniciusCMB , I reviewed the architecture. This is not deprecated, it is actually a very important feature for simulating realistic avionics. The challenge is that you need to calculate u_dot(t, y) inside the event loop to get the acceleration. I have updated the issue description with the technical details.

In case you want to read a more detailed description, here it goes:

This is a **critical** feature for advanced users. It is definitely **still necessary**.
Here is the breakdown of why, how to solve it, and the updated issue description.

### 1\. Is this still necessary? **YES.**

**Why?**
Real-world rocket avionics (flight computers) rarely use "altitude" or "vertical velocity" directly to detect events because barometers are noisy and GPS is slow.
Most avionics trigger events based on **Accelerometer** data (IMU):

  * **Liftoff detection:** High positive acceleration.
  * **Burnout detection:** Sudden drop in acceleration.
  * **Apogee detection:** 0 velocity (integrated from acceleration).

Currently, RocketPy forces users to trigger parachutes based on *state variables* (Position `z`, Velocity `vz`) or Pressure. This is "God Mode" simulation. To simulate **realistic avionics logic**, the user needs access to `acceleration` (and ideally, noisy acceleration) inside the trigger function.

### 2\. How can we do this?

**The Problem:**
RocketPy uses `scipy.integrate.solve_ivp`.

  * The solver passes `(t, y)` (Time and State Vector) to the Event/Trigger function.
  * Acceleration is NOT in the State Vector `y` (which is `[x, v, ...]`).
  * Acceleration is the *derivative* `y'` (or `u_dot`).

**The Solution:**
To get acceleration inside the trigger, we must calculate the derivative **inside the event function**.

1.  **Current Flow:**
    `Trigger(pressure, state_vector)`
2.  **New Flow:**
    Inside the internal wrapper that RocketPy uses for events:
    ```python
    def parachute_trigger_wrapper(t, y):
        # 1. Calculate the physics derivative at this exact moment
        # This gives us [vx, vy, vz, ax, ay, az, ...]
        u_dot = flight.u_dot(t, y) 
        
        # 2. Extract acceleration
        acc_z = u_dot[5] # (assuming index 5 is az)
        
        # 3. (Optional) Add Sensor Noise here if strictly required
        
        # 4. Call the user's trigger function with extra data
        return user_trigger_function(pressure, y, acceleration=acc_z)
    ```

*Trade-off:* This adds computational cost (physics is calculated twice: once for the step, once for the event check). However, this is the only accurate way to do it.