RocketPy-Team/RocketPy #158: Distances vs. Positions...

giovaniceotto, 2022-03-17:
**Is your feature request related to a problem? Please describe.**

RocketPy's Rocket class arguments can be quite confusing, especially distances (`distanceRocketNozzle`, `distanceRocketPropellant`, `distanceToCM`).

**Describe the solution you'd like**

A new way of entering the position of interest points. It would be extremely nice if you could choose the origin, instead of it being fixed in the center of dry mass.

Furthermore, the naming of center of mass should change to center of dry mass.

**Additional context**

Measuring things using the nose cone as the origin may be easier sometimes!

Gui-FernandesBR, 2022-03-21:
@MateusStano do you accept this challenge? 

(Of course we will help you along this journey...)

MateusStano, 2022-07-06:
Challenge accepted and completed!! See --> #195

closed 2022-07-06 by MateusStano