RocketPy-Team/RocketPy #161: Parachute inertial impact force

Gui-FernandesBR, 2022-03-21:
Calculate the Parachute deployment impact forces, 

I believe it only depends on 
1. Rocket Freestream Speed at the instant of deployment
2. Parachute CdS
3. A Force Coefficient based on the parachute geometry (i.e. if it is ellipsoid, circular, with or without vent, etc.)

@FranzYuri do you think we could implement it on utilities.py, parachute.py or flight.py file? I heve in mind that you were studying this for a while last year, right?

I have some literatures on this topics if you need, just let me know

FranzYuri, 2022-06-16:
@Gui-FernandesBR sure! I started the development at utilities.py, using an estimation for the peak opening force available in Knacke's Parachute Recovery Systems Design Manual page 5-50.  Curently, the idea is to make a simple function, using as input only the velocoty of the rocket in the moment of the ejection, height, possibly the environment(to get the air velocity) and the opening-force-reduction factor (I am studying the possibility of, instead of this, using the mass of the rocket).

FranzYuri, 2022-06-16:
[image]