RocketPy-Team/RocketPy #162: Implement MRS method on RocketPy!

Gui-FernandesBR, 2022-03-21:
MRS is an awesome method which is available on https://github.com/Projeto-Jupiter/RocketPaper/tree/main/Multivariate%20Rejection%20Sampling , it would be easy to be implemented on RocketPy's repository and for a first moment I think utilities.py would be the best home for it (of course it can be discussed).

**It's important that we bring MRS to this repo so it doesn't become forgotten** 

Also, a quick introduction to MRS powerfull capacity would be welcome as well

giovaniceotto, 2022-03-22:
Should a more integrated version of the dispersion analysis be implemented before MRS?

Gui-FernandesBR, 2022-03-22:
I believe @MateusStano has something to say about it

giovaniceotto, 2022-03-26:
Can't wait to hear more from @MateusStano!