RocketPy-Team/RocketPy #164: What if we rearrange postProcess method?

Gui-FernandesBR, 2022-03-28:
**Is your feature request related to a problem? Please describe.**

Sometimes it feels like the Flight.postProcess method is draining our time away. Everytime you need to evaluate one specific function (e.g. Flight.R2) you have to run the simulation setting postProcess() in order to catch the desired values.
The problem is that sometimes you don't wanna post process every single attribute, but only a set of data instead.

My suggestion here is that we make with postProcess something similar to what happened on allInfo() few years ago... We could agregate some attributes of postProcess together and . All in all, the old postProcess will be trasnformed in a function that only need to call the other methods, this way not introducing breaking cahnges

**Questions**
What do you guys think about such ideia? Do you have suggestions before we start coding?
Below I give u an example of what would be the new postProcess structure.

The only atention point is that some functions depends on others, meaning that, for insstance, if you want to evaluate "SpecialFunction" is will need to evaluate "postProcessAeroForces" together.
[image]

giovaniceotto, 2022-03-28:
Yes please!!! We really need to refactor the `Flight.postProcess` method urgently. I like your idea of breaking it down by topics, similar to the `Flight.allInfo` method.

I'll add one more idea to the table: [Python's property(): Add Managed Attributes to Your Classes](https://realpython.com/python-property/ )

Or, if all you need a quick intro to property decorators: [Python Property Decorator - @property](https://www.tutorialsteacher.com/python/property-decorator)

Using the property decorator, we could create a managed attribute for each post process variable and really calculate them only when they are called, independently.

Gui-FernandesBR, 2022-11-03:
Solved by #249

closed 2022-11-03 by Gui-FernandesBR