RocketPy-Team/RocketPy #205: RuntimeError: Unable to load latest weather data for GFS

git-akihakune, 2022-07-28:
**Describe the bug**

Running the code in [Colab Notebook preview](https://colab.research.google.com/github/giovaniceotto/rocketpy/blob/master/docs/notebooks/getting_started_colab.ipynb) led to an error.

```python
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
[<ipython-input-17-71fdef7bdc29>](https://localhost:8080/#) in <module>()
----> 1 Env.setAtmosphericModel(type="Forecast", file="GFS")

[/usr/local/lib/python3.7/dist-packages/rocketpy/Environment.py](https://localhost:8080/#) in setAtmosphericModel(self, type, file, dictionary, pressure, temperature, wind_u, wind_v)
    935                 if not success:
    936                     raise RuntimeError(
--> 937                         "Unable to load latest weather data for GFS through " + file
    938                     )
    939             elif file == "FV3":

RuntimeError: Unable to load latest weather data for GFS through https://nomads.ncep.noaa.gov/dods/gfs_0p25/gfs20220717/gfs_0p25_06z
```

**To Reproduce**

Just run the default code in the preview.

**Expected behavior**

It's expected to run the demo smoothly.

**Screenshots**

[image]

**Additional context**

Checking the [GrADS Data Server](https://nomads.ncep.noaa.gov/dods/gfs_0p25), indeed, I'm currently not seeing any "gfs20220717" dir. Maybe fixing the link can solve this issue.

giovaniceotto, 2022-07-28:
@git-akihakune Thanks for letting us know that you are having an issue. We recently became aware of this and it seems to be a problem relating the netCDF4 library and Colab.

We have not been able to reproduce the error locally so far. I will perform further investigations and let you know.

giovaniceotto, 2022-07-28:
I have just confirmed that running on a linux server (GitHub Codespaces) works properly. It really seems to be a problem specific to Google Colab.

Furthermore, I have tested a simple netCDF4 example using Colab and  also got an error. You can see the example here: https://colab.research.google.com/drive/17juZulxoz8sahoDJVuU4uOMSnBC_xXDA?usp=sharing

On other systems, the example runs fine.

This maybe a related issue: https://github.com/Unidata/netcdf4-python/issues/753

giovaniceotto, 2022-07-28:
I have created a new issue on netCDF4's repo for us to track while we investigate this: https://github.com/Unidata/netcdf4-python/issues/1179

giovaniceotto, 2022-07-31:
@git-akihakune I found a quick workaraound: installing an earlier version of `netCDF4` solves the problem for me.

You can try by running `!pip install "netCDF4<1.6.0"` in Colab and checking if it also solves your issue.

git-akihakune, 2022-08-01:
The workaround was great. I've tested on a few different Colab notebooks and no error was reported.

As for this issue, I suggest leaving it open until the problem is entirely solved.

Gui-FernandesBR, 2022-08-01:
> As for this issue, I suggest leaving it open until the problem is entirely solved.

Agreed!!

OliverKillane, 2022-10-19:
```python
Env.setAtmosphericModel(type='Forecast', file='GFS', dictionary='NOAA')
```

Having the same problem here:

With  netCDF4-1.6.1 the same error (date in url different)

With netcdf4-1.5.8 it works

Gui-FernandesBR, 2022-10-24:
Everything seems to be working now, 

@giovaniceotto shall we close this one or wait for a resolution on https://github.com/Unidata/netcdf4-python/issues/753 ?

imo we can already close

giovaniceotto, 2022-10-25:
The issue we are tracking is actually https://github.com/Unidata/netcdf4-python/issues/1179

Yes, let's leave it open until the problem is fully resolved.

Gui-FernandesBR, 2023-06-09:
> The issue we are tracking is actually [Unidata/netcdf4-python#1179](https://github.com/Unidata/netcdf4-python/issues/1179)
> 
> Yes, let's leave it open until the problem is fully resolved.

It seems like the referred issue is going to be closed in the short-term future, just waiting for someone else to confirm that the `netcdf4==1.6.4` version addressed the problem. I can confirm it is running without errors on colab (which now has default python 3.10)

Gui-FernandesBR, 2023-10-09:
Solved, everything is running on `rocketpy==1.0.0` and above

closed 2023-10-09 by Gui-FernandesBR