RocketPy-Team/RocketPy #207: type cast issue with running rocketpy

Ashur59, 2022-08-04:
I would like to create a trajectory path for a rocket utilizing `rocketpy`. Here is a sample code:

    from rocketpy import Environment, Rocket, SolidMotor, Flight

    # create an Environment object
    Env = Environment(
        railLength=5.2,
        latitude=32.990254,
        longitude=-106.974998,
        elevation=1400,
        date=(2020, 3, 4, 12)  # Tomorrow's date in year, month, day, hour UTC format
    )
    Env.setAtmosphericModel(type='Forecast', file='GFS')

And here is the error message I am getting:

    File "c:\Users\username\Desktop\Some Project\trajectory_creator.py", line 20, in <module>
        Env.setAtmosphericModel(type='Forecast', file='GFS')
      File "C:\ProgramData\Anaconda3\envs\tf\lib\site-packages\rocketpy\Environment.py", line 873, in setAtmosphericModel
        self.processForecastReanalysis(file, dictionary)
      File "C:\ProgramData\Anaconda3\envs\tf\lib\site-packages\rocketpy\Environment.py", line 1710, in processForecastReanalysis       
        fileTimeDate = netCDF4.num2date(timeArray[timeIndex],
      File "src\cftime\_cftime.pyx", line 584, in cftime._cftime.num2date
      File "src\cftime\_cftime.pyx", line 382, in cftime._cftime.cast_to_int
    TypeError: '<' not supported between instances of 'numpy.longdouble' and 'int'

I am not sure how to resolve the issue. I installed the package using pip install rocketpy into my tensorflow environment on a conda platform.

giovaniceotto, 2022-08-05:
Thanks for raising the issue @Ashur59.

I have tried to reproduce it in my local machine and in a Google Colab environment with no success, however.

I had to adjust the dates, however, to get it working. Forecasts can only be used with future dates. Here is the code I used:

```python
from rocketpy import Environment, Rocket, SolidMotor, Flight

# create an Environment object
Env = Environment(
    railLength=5.2,
    latitude=32.990254,
    longitude=-106.974998,
    elevation=1400,
    date=(2022, 8, 5, 12)  # Tomorrow's date in year, month, day, hour UTC format
)
Env.setAtmosphericModel(type='Forecast', file='GFS')
```

And here is the Colab notebook for it: https://colab.research.google.com/drive/1xQ7sua6Aq4S6C1D0xocNKyvuFCoCPaDD?usp=sharing

Let me know more details about your enviornment to see if I can detect anything.

Ashur59, 2022-08-05:
I am running the code on a visual studio code on windows 10 and regardless of the choice of "date" that needs to be future, it is still not working with the said error message:

```
oc_open: server error retrieving url: code=0 message="/gfs_0p25/gfs20220805/gfs_0p25_12z is not an available dataset"Traceback (most recent call last):
  File "c:\Users\user_name\Desktop\Projects\trajectory_creator.py", line 20, in <module>
    Env.setAtmosphericModel(type='Forecast', file='GFS')
  File "C:\ProgramData\Anaconda3\envs\tf\lib\site-packages\rocketpy\Environment.py", line 873, in setAtmosphericModel       
    self.processForecastReanalysis(file, dictionary)
  File "C:\ProgramData\Anaconda3\envs\tf\lib\site-packages\rocketpy\Environment.py", line 1710, in processForecastReanalysis
    fileTimeDate = netCDF4.num2date(timeArray[timeIndex],
  File "src\cftime\_cftime.pyx", line 584, in cftime._cftime.num2date
  File "src\cftime\_cftime.pyx", line 382, in cftime._cftime.cast_to_int
TypeError: '<' not supported between instances of 'numpy.longdouble' and 'int'
```

giovaniceotto, 2022-08-05:
@Ashur59, take a look at issue #205

giovaniceotto, 2022-08-07:
@Ashur59, did installing an earlier version of `netCDF4` help with this issue?

Ashur59, 2022-08-08:
Hi Giovaniceotto,
I did try to run the command pip install "netCDF4<1.6.0" but it still did not fix the issue. I will update you if I could fix it somehow.

Ashur59, 2022-08-08:
@giovaniceotto, running `pip install --upgrade --force-reinstall netCDF4` fixed the dependency of the CDF4 on other python packages and now the code runs smoothly. I think some of the other python packages had a conflict with the latest version of CDF4.

giovaniceotto, 2022-08-08:
That is nice to hear @Ashur59! I will close this issue then.

closed 2022-08-08 by giovaniceotto