RocketPy-Team/RocketPy #242: ENH: Automate dispersion Outputs extraction to .csv

Gui-FernandesBR, 2022-09-24:
**Is your feature request related to a problem? Please describe.**

As suggested by @MateusStano  at #229 .

**Describe the solution you'd like**

We should easily save the disrpersion outputs into an external file.
It is important when we have to simulate several rockets over different scenarios and perform a quick comparison.

Gui-FernandesBR, 2023-04-17:
Perhaps they [the outputs from the Monte Carlo simlations] should be exported to ;json files instead of .txt (current) or .csv (proposed)

GMinoruy, 2025-12-04:
Hey, I am that user on Discord that asked where to be assigned issues, this is the one I was considering for the selective process can I work on it?