RocketPy-Team/RocketPy #261: SyntaxError: current version fails to run on Python releases earlier than 3.8

MrGribel, 2022-10-22:
**Describe the bug**
When importing the rocketpy module on python versions earlier than 3.8, a syntax error occurs due to a unpacking operator present in a return statement (a feature introduced recently) in the EnvironmentAnalysis class. 

**To Reproduce**

Try running the current PyPI rocketpy version on any python versions earlier than 3.8.

**Expected behavior**

Package initialization should be successful on all python versions described on the requirements docs (3.6+), or requirements should be changed.

**Screenshots**

[image]

**Additional context**

Google Colab online environment defaults to Python version 3.7, so this issue breaks all the current online RocketPy Jupyter notebooks. This requires the user to run on an older version (before the merge of EnvironmentAnalysis), or update their Colab kernel with a more recent python version.

Gui-FernandesBR, 2022-10-22:
By "current version", understand v0.12.0 (https://pypi.org/project/rocketpy/0.12.0/).

This issue was introduced on v0.11.0, which means that the newest version that is not dependent of python 3.8 is the 0.10.0 (https://pypi.org/project/rocketpy/0.10.0/)

Gui-FernandesBR, 2022-10-24:
Solved!

[image]

closed 2022-10-24 by Gui-FernandesBR