RocketPy-Team/RocketPy #265: BUG: Unable to setup RocketPy using python 3.11

Gui-FernandesBR, 2022-11-02:
**Describe the bug**

When used the most recent [python 3.11 version](https://docs.python.org/3/whatsnew/3.11.html), rocketpy cannot be properly installed, it seems that the interpreter is currently having issues when installing the netCDF4 library. More specifically, a `error: metadata-generation-failed` is raised.

**To Reproduce**

Initialize a fresh environment of python and ensure you are using python 3.11.0 version.

**Expected behavior**

I want to see something like this on my terminal after setting up rocketpy package:
`Successfully installed certifi-2022.9.24 cffi-1.15.1 cftime-1.6.2 charset-normalizer-2.1.1 contourpy-1.0.6 cycler-0.11.0 fonttools-4.38.0 h3-3.7.4 idna-3.4 jsonpickle-2.2.0 kiwisolver-1.4.4 matplotlib-3.6.1 netCDF4-1.6.1 numpy-1.23.4 packaging-21.3 pillow-9.3.0 pycparser-2.21 pyparsing-3.0.9 python-dateutil-2.8.2 pytz-2022.6 requests-2.28.1 rocketpy-0.12.0 scipy-1.9.3 setuptools-65.5.0 simplekml-1.3.6 six-1.16.0 timezonefinder-6.1.6 urllib3-1.26.12 windrose-1.8.0`

**Screenshots**

**Trying to install rocketpy with python 3.11:**
[image]
[image]

**Now, when trying on python 3.10:**
[image]
[image]
[image]

Gui-FernandesBR, 2022-11-02:
Does anyone have any idea on how to solve it?

@giovaniceotto @MateusStano @phmbressan @MrGribel

phmbressan, 2022-11-02:
I was able to reproduce the issue. It seems to me that both `netCDF4` and its dependency `h5py` lack wheels for `python3.11.0`.

I don't have yet a temporary fix for the problem, it is likely that in the upcoming weeks they will launch the wheels for `python3.11.0`. I will link here related issues from these packages:

[netCDF4 3.11 wheel](https://github.com/Unidata/netcdf4-python/issues/1204)

[h5py PR 3.11 wheel](https://github.com/h5py/h5py/pull/2165)

I believe that it is possible to circumvent this problem by building both packages from source, but I had some difficulties with package dependencies.

giovaniceotto, 2022-11-03:
Besides building both packages from source, which can be quite a challenge, especially on Windows, another solution that I have found is by relaying on @conda-forge.

Conda-forge has already started the effort to support Python 3.11, and the status of the migration can be followed through this dedicated page: https://conda-forge.org/status/#python311

Both `h5py` and `netCDF4` have already been migrated, which means that they can be installed in a conda environment with python 3.11.

Here is a recipe of how to do it:

1. Install [Anaconda](https://www.anaconda.com/products/distribution) or [Mini-Conda](https://docs.conda.io/en/latest/miniconda.html)
2. Start the Anaconda Prompt and create a new environment to mess with: `conda create -n py311`.
3. Activate the environment: `conda activate py311`.
4. Install Python 3.11: `conda install -c conda-forge python` .
5. Install h5py and netCDF4: `conda install -c conda-forge h5py netCDF4`
6. Install RocketPy using pip: `pip install rocketpy`

The result is as follows:
[image]

Should work on any OS.

phmbressan, 2023-02-19:
Update: it seems now that `h5py` have released `python3.11.0` wheels in its newest version. 

I was able to `pip install rocketpy` on `python3.11.0`, however newer versions still gives an error.

Gui-FernandesBR, 2023-04-15:
Yeah finally!!

Lucas-KB, 2023-05-20:
Today I tried to install rocketpy and I realized it still fails running `pip install rocketpy`. I believe it is related to this issue.

I have tested it on Fedora, Debian and Alpine linux distributions using Python 3.11 and none of them worked. The installation attempt logs are reproduced below:

Fedora 38 Workstation:
- Python version: 3.11.3
- Pip version: 22.3.1

```
Defaulting to user installation because normal site-packages is not writeable
Collecting rocketpy
  Using cached rocketpy-0.13.1-py3-none-any.whl (129 kB)
Requirement already satisfied: numpy>=1.0 in /usr/lib64/python3.11/site-packages (from rocketpy) (1.24.3)
Requirement already satisfied: scipy>=1.0 in ./.local/lib/python3.11/site-packages (from rocketpy) (1.10.1)
Requirement already satisfied: matplotlib>=3.0 in ./.local/lib/python3.11/site-packages (from rocketpy) (3.7.1)
Collecting netCDF4<1.6,>=1.4
  Using cached netCDF4-1.5.8.tar.gz (767 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [34 lines of output]
      Package hdf5 was not found in the pkg-config search path.
      Perhaps you should add the directory containing `hdf5.pc'
      to the PKG_CONFIG_PATH environment variable
      Package 'hdf5', required by 'virtual:world', not found
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-huy4afe8/netcdf4_c601c881fceb4d62ab378fc02e0dce34/setup.py", line 435, in <module>
          raise ValueError('did not find netCDF version 4 headers')
      ValueError: did not find netCDF version 4 headers
      reading from setup.cfg...
      
          HDF5_DIR environment variable not set, checking some standard locations ..
      checking /home/lucas/include ...
      hdf5 headers not found in /home/lucas/include
      checking /usr/local/include ...
      hdf5 headers not found in /usr/local/include
      checking /sw/include ...
      hdf5 headers not found in /sw/include
      checking /opt/include ...
      hdf5 headers not found in /opt/include
      checking /opt/local/include ...
      hdf5 headers not found in /opt/local/include
      checking /usr/include ...
      HDF5 library version: 1.12.1 headers found in /usr/include
      HDF5 library version: 1.12.1 found in /usr
      
      NETCDF4_DIR environment variable not set, checking standard locations..
      checking /home/lucas ...
      checking /usr/local ...
      checking /sw ...
      checking /opt ...
      checking /opt/local ...
      checking /usr ...
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

Debian Bulseye:
[... 413 lines cut ...]
                raise RuntimeError(
            RuntimeError: Broken toolchain: cannot link a simple C program.
            [end of output]
      
        note: This error originates from a subprocess, and is likely not a problem with pip.
        ERROR: Failed building wheel for numpy
      Failed to build numpy
      ERROR: Could not build wheels for numpy, which is required to install pyproject.toml-based projects
      
      [notice] A new release of pip available: 22.3.1 -> 23.1.2
      [notice] To update, run: pip install --upgrade pip
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

[notice] A new release of pip available: 22.3.1 -> 23.1.2
[notice] To update, run: pip install --upgrade pip
```

Gui-FernandesBR, 2023-06-09:
Many thanks @Lucas-KB for pointing that out.
It seems that netcdf4 cannot be built in your environment based on the requirements given in the setup.py file.

Could you please try to build rocketpy again using the `netcdf4==1.6.4` version? This should fix the problem.
That's a new version that was released a few days ago.

Gui-FernandesBR, 2023-10-15:
RocketPy has been running flawlessly on Python 3.11 for the past few months.
Closing this issue for now.

Let's see what happens with the recent Python 3.12 release.

closed 2023-10-15 by Gui-FernandesBR