RocketPy-Team/RocketPy #272: Supersonic Aerodynamics

giovaniceotto, 2022-11-05:
RocketPy implements Barrowman's equations for subsonic compressible flow. As more and more flights achieve the supersonic regime, it is crucial that our aerodynamic model stays accurate at such conditions.

Therefore, this issue requests the implementation of Barrowman's equations considering supersonic flow.

## Requirements

1. Have a reasonably accurate model of normal aerodynamic forces for subsonic, transonic and supersonic flight.

2. Be able to evaluate the stability margin as a function of Mach number.

## Proposed Milestone

**v1.1.0**

robertobilbao, 2023-07-31:
Is there any method to simulate the trayectory of supersonic Rockets on RocketPy?

Gui-FernandesBR, 2024-11-02:
Regarding the 2 required features of this issue:

> Have a reasonably accurate model of normal aerodynamic forces for subsonic, transonic and supersonic flight.

We recently released the GenericSurfaces, which allowed for the import of completly customized aero curves! https://github.com/RocketPy-Team/RocketPy/pull/680

> Be able to evaluate the stability margin as a function of Mach number.

This was solved one year ago!
[image]

With that being said, I'm marking this issue as solved.

closed 2024-11-02 by Gui-FernandesBR