RocketPy-Team/RocketPy #284: Built-in compare flights

Gui-FernandesBR, 2022-11-12:
**Is your feature request related to a problem? Please describe.**

More and more teams keeping asking us about RocketPy performance. MOre specifically, they usually want to know how would RocketPy simulation performs when considering their own rockets.

**Describe the solution you'd like**

Simple but flexible, create methods that receive data from other simulators or from avionics logs. And then it must: 
- [ ] Compare, plot, report to the user.

**Additional context**

See [RocketPaper](https://github.com/RocketPy-Team/RocketPaper) repository to find some (perhaps) useful codes with the same porpouse.

Gui-FernandesBR, 2022-11-12:
Altough not exactly the same problem, the #252 is kind related to this one

Monta120, 2025-11-27:
Hi! I’d like to contribute to Issue #284. Since FlightDataImporter already handles importing external flight data, I can implement the next layer: a FlightComparator class that takes a RocketPy Flight and imported flight data and automatically generates comparison plots (altitude, velocity, acceleration, attitude) and summary error metrics (RMS error, max deviation, etc.).

This would provide the built-in “compare flights” functionality requested in the issue. Let me know if this aligns with what you expect.

closed 2025-12-02 by Gui-FernandesBR