RocketPy-Team/RocketPy #297: addTrapezoidalFins docs not rendering corretly

Gui-FernandesBR, 2022-11-14:
**Describe the bug**

When accessing the documentation of the `Rocket.addTrapezoidalFins` method. The docs that are built are not following the correct format.

**To Reproduce**

Just access the following link: (https://docs.rocketpy.org/en/latest/reference/classes/Rocket.html?highlight=addtrapezo#rocketpy.Rocket.Rocket.addTrapezoidalFins)

**Expected behavior**

Something like this whoud happen:
[image]

**Screenshots**

[image]

**Additional context**

This issue was reported by an user in our discord server. I'm saving here so we don't forget it!

giovaniceotto, 2022-11-14:
May be useful to prevent future errors similar to this: https://stackoverflow.com/questions/33896196/how-to-test-the-validity-of-your-sphinx-documentation

Gui-FernandesBR, 2025-08-15:
The readthedocs platform already deploys new versions of our docs on every push. In case of failure, we get notified.
Therefore I'm closing this issue as not planned, given that our problem has been solved using other approach.

closed 2025-08-15 by Gui-FernandesBR