RocketPy-Team/RocketPy #309: Auto burn out time

giovaniceotto, 2022-12-03:
**Is your feature request related to a problem? Please describe.**

It can be quite redundant to specify a burn out time when defining motors in RocketPy. Since most motors are defined using .csv or .eng thrust curves, the burn out time can be easily retrieved automatically by the code.

**Describe the solution you'd like**

I would like the burn out time attribute to be refactored into the following behaviour:

```
burn_time: float, tuple of float, optional
    Motor's burn time.
    If a float is given, the burn time is assumed to be between 0 and the given float, in seconds.
    If a tuple of float is given, the burn time is assumed to be between the first and second elements of the tuple, in seconds.
    If not specified, automatically sourced as the range between the first- and last-time step of the motor's thrust curve. This can only be used if the motor's thrust is defined by a list of points, such as a .csv file, a .eng file or a Function instance whose source is a list.
```

**Additional context**

It is often hard to explain to users why they need to supply a burn time. It is quite confusing to them since this information appears redundant, and it often is. However, eliminating this argument all together does not seem like a good option in my opinion, since it would make it impossible to defined motors with a simple constant Function thrust source, quite useful when experimenting with rockets.

Gui-FernandesBR, 2022-12-03:
I agree with the suggestion, let's move forward with that.

@MateusStano please add in your roadmap.

Gui-FernandesBR, 2023-06-25:
Fixed with #337 ✅

closed 2023-06-25 by Gui-FernandesBR