RocketPy-Team/RocketPy #310: Function filters

giovaniceotto, 2022-12-03:
**Is your feature request related to a problem? Please describe.**

Often, experimental data is loaded in RocketPy, and it can be quite noisy. This can make simulations behave quite unpredictably, especially when the noisy data is differentiated inside the code.

**Describe the solution you'd like**

 It would be helpful if the Function class had some methods to be able to quickly apply simple filters, such as moving average, high-pass, low-pass and band-pass.

**Additional context**

For some liquid motor related classes, the input data is differentiated to compute relevant parameters. When the input data is noisy, it's derivative is severely impacted to the point where it can't really be used reliably.

kalounis, 2023-11-15:
Hello, is it possible to be assign on the issue pls? Thank you!

Gui-FernandesBR, 2023-11-15:
> Hello, is it possible to be assign on the issue pls? Thank you!

Done!

Let us know if you need any help on that, please.

XxWillyxX, 2023-11-28:
Hello, is it possible to be assign on the issuetoo pls? Thank you!

Gui-FernandesBR, 2024-01-12:
I'm okay with closing this issue since the low_pass filter function was added. 
Other types of filters can be done under specific requirements in the future.

closed 2024-01-18 by phmbressan