RocketPy-Team/RocketPy #314: Significant Dependency Issue: Timezonefinder does not support Python 3.7

giovaniceotto, 2022-12-31:
**Describe the bug**

[Timezonefinder](https://github.com/jannikmi/timezonefinder) no longer supports python 3.7. This is a significant issue since RocketPy officially supports Python 3.7 but depends on `timezonefinder`.

**To Reproduce**

Create an environment with Python 3.7.9 on Windows and attempt to install RocketPy. The following error should occur (tested locally and via GitHub Actions):

```shell
Best match: timezonefinder 6.1.9
Processing timezonefinder-6.1.9.tar.gz
error: Couldn't find a setup script in C:\Users\ghceo\AppData\Local\Temp\easy_install-byqibg5q\timezonefinder-6.1.9.tar.gz
```

**Expected behavior**

RocketPy should install without any issues, but it fails since a required library cannot be installed.

**Additional context**

This started to be a problem only recently. Perhaps requiring an earlier version of `timezonefinder` would solve the issue. However, this is dangerous for many reasons since we can end up with out of dated time zone delimitations.

giovaniceotto, 2022-12-31:
Did a little digging and found out that `timezonefinder` dropped support for version 3.7 once Numpy started supporting versions 3.8 and above.

I believe we should do the same for release v.1.0.0: drop support for Python 3.7, support Python 3.8 - Python 3.11.

Gui-FernandesBR, 2022-12-31:
If numpy dropped 3.7, I agree dropping as well.

On the other hand, why not just dropping timezonefinder then? This is a not so light weight library, and probably there are other options available.

Gui-FernandesBR, 2022-12-31:
I think @phmbressan may have an opinion as well

giovaniceotto, 2022-12-31:
> On the other hand, why not just dropping timezonefinder then? This is a not so light weight library, and probably there are other options available.

There is no other viable alternative to `timezonefinder` available, in my opinion. If we drop it, we will drop the feature that automatically detects the timezone based on latitude and longitude.

giovaniceotto, 2022-12-31:
To add to the discussion of dropping this library, or making it optional, it is currently not implemented in the `Environment` class. It is only used in the `EnvrionmentAnalysis` class. In fact, it is only used briefly in one method: https://github.com/RocketPy-Team/RocketPy/blob/master/rocketpy/EnvironmentAnalysis.py#L346

It can easily be made an optional dependecy!

giovaniceotto, 2022-12-31:
I went ahead and created #315. It does not solve the discussion of dropping support for Python 3.7 or not. It only makes the `timezonefinder` package requirement optional, which, in my honest opinion, should be the case regardless since it is not necessary to run the basic functionality of RocketPy.

MateusStano, 2023-01-02:
> I believe we should do the same for release v.1.0.0: drop support for Python 3.7, support Python 3.8 - Python 3.11.

It definitely makes sense to drop Python 3.7 for v1.0 

Are there any other libraries that should be made optional? I just checked and there are a few other EnvironmentAnalysis libraries that are required for installation, i.e. windrose, jsonpickle and ipywidgets. 

I'd suggest we open a PR to remove the Python 3.7 dependency and make these other libraries optional as well

giovaniceotto, 2023-01-10:
Matplotlib has dropped support for Python 3.7 in its latest version as well.

Gui-FernandesBR, 2023-05-26:
> It definitely makes sense to drop Python 3.7 for v1.0
>
> Are there any other libraries that should be made optional? I just checked and there are a few other EnvironmentAnalysis libraries that are [...]

I recently created the PRs #365 and #368, which are going to drop the support for python version 3.7 in rocketpy and reduce the number of mandatory dependencies, respectively.

After merging these two PRs, I believe we are going to be able to close this one, finally.

MateusStano, 2023-06-20:
Solved in #365 and #368

closed 2023-06-20 by MateusStano