RocketPy-Team/RocketPy #318: "StandardAtmosphere" environment pressure calculation

MateusStano, 2023-01-02:
## **Describe the bug**

Looking at how the pressure function is calculated for `"StandardAtmosphere"` I found that lines 2672 and 2673 are incoherent:

https://github.com/RocketPy-Team/RocketPy/blob/48d69495bdb93fe5709f4c62e8fa8f932b8e1df1/rocketpy/Environment.py#L2670-L2707

The same thing happens in 2654 and 2655

https://github.com/RocketPy-Team/RocketPy/blob/48d69495bdb93fe5709f4c62e8fa8f932b8e1df1/rocketpy/Environment.py#L2652-L2656

**Additional context**

I believe @giovaniceotto might know what is going on here.

This issue continues the discussion on #313.

Gui-FernandesBR, 2023-05-26:
Please notice this problem also happen in line 2815 of the same file.

Gui-FernandesBR, 2023-05-26:
Just opened the #369 PR, it is meant to be a quick solution for this issue.

Let's go!

Gui-FernandesBR, 2023-05-29:
Solved in #369 ✅

closed 2023-05-29 by Gui-FernandesBR