RocketPy-Team/RocketPy #350: Stability Margin with Compressibility Factor

giovaniceotto, 2023-04-10:
**Is your feature request related to a problem? Please describe.**

Currently, the `Flight` class has a property named `Flight.staticMargin`, which returns the static margin as a function of time. However, since it is the static margin, it always assumes `Mach=0.0`, and, thus, the [Prandtl-Glauert compressibility correction factor](https://en.wikipedia.org/wiki/Prandtl%E2%80%93Glauert_transformation) $beta$ is always 0. 

**Describe the solution you'd like**

For the `Rocket` class:
- `Rocket.initialStabilityMargin(mach)`: a method which computes the initial stability margin given a Mach number.

For the `Flight` class:
- `Flight.stabilityMargin()`: a property which returns the stability margin, as a Function of time, for the entire flight, accounting for Mach numbers different from 0, and changing center of mass position.

**Additional context**

This is a common request from OpenRocket users.

Gui-FernandesBR, 2023-05-23:
Hey, this is a common request from RocketPy users, let's increase this priority right after release v1.0.0

Please @MateusStano  👨‍💻

FYI,
[image]

Gui-FernandesBR, 2023-10-13:
Seems like the issue statement was solved by the #377 
We still have some enhancements to do regarding the stability margin, tho.

closed 2023-10-13 by Gui-FernandesBR