RocketPy-Team/RocketPy #371: BUG: EnvironmentAnalysis altitudes ASL vs AGL

Gui-FernandesBR, 2023-05-29:
**Describe the bug**

I noticed this bug while reading the EnvironmentAnalysis class again. If you read [this line](https://github.com/RocketPy-Team/RocketPy/blob/171e61f9d7705dd89cafc31ee96602eda1b8a726/rocketpy/EnvironmentAnalysis.py#LL666C3-L666C3) at master branch, you're gonna notice that the altitudes arrays are a little bit confused between ASL and AGL values.

It seems that the [this commit](https://github.com/RocketPy-Team/RocketPy/commit/ad56fe934a061e4f4887215cd97521c195c0f239) didn't adjust all the occurrences of the `heightAboveSeaLevelArray` variable. This makes some variables to be referred to altitudes AGL while having an ASL in the name/dimensions.

It seems that the 
Sadly, I am right, this should evidence a major bug on the EnvironmentAnalysis example.

**To Reproduce**

It is not about reproducing, instead, try to simply read the method `parsePressureLevelData` and understand whether it is a bug or not. In my opinion, it should be easy to identify that few Function are wrong. Just follow the screenshot below.

**Expected behavior**

Same as above.

**Screenshots**

[image]

**Additional context**

Tagging @giovaniceotto as he was the responsible for the mentioned commit, however, I believe that anyone else could also take a look into it.

Gui-FernandesBR, 2023-05-29:
Step by step solution here:

- [ ] First of all, we need to confirm it is a BUG
- [ ] Choose whether to use ASL or AGL altitudes from now on
- [ ] Submit a PR fixing the wrong assignments

I need some help on the first task here (@MateusStano @giovaniceotto @phmbressan ), 

and then I can be responsible for fixing it XD

MateusStano, 2023-06-21:
> * First of all, we need to confirm it is a BUG

It definitely is a bug

> * Choose whether to use ASL or AGL altitudes from now on

We should use AGL since it is easier to understand and seems to be the original intention of the code

closed 2023-09-07 by Gui-FernandesBR