RocketPy-Team/RocketPy #379: Hybrid Motor Support

ryan3507, 2023-06-21:
I saw on the website that hybrid motor support is coming within the next few weeks. Any idea when that will be coming? Thanks and great work!

ryan3507, 2023-06-22:
Okay thank you!

ryan3507, 2023-06-22:
One more question. Did you guys ever test your model with the Saturn V or any other missions to see how accurate it was? Thank you again!

Gui-FernandesBR, 2023-06-23:
> One more question. Did you guys ever test your model with the Saturn V or any other missions to see how accurate it was? Thank you again!

At the best of my knowledge, I don't think someone already tried that.
But I would be really interested to see a simulation like that! 
Btw orbital simulations is something still not so explored with RocketPy, and we do think about some modifications that could enhance simulations of this kind. See https://github.com/RocketPy-Team/RocketPy/issues/277

Feel free to ask help if you want to try it from your side :)

ryan3507, 2023-06-29:
> *External Email*

Okay cool. Would you be willing to help me try to simulate that and see the
results? Thank you again for all the help!

closed 2023-07-04 by Gui-FernandesBR