RocketPy-Team/RocketPy #411: BUG: the rocket does not take off the rail

Gui-FernandesBR, 2023-09-16:
**Describe the bug**

- I did everything correctly when creating my Flight object, but for some reason the rocket doesn't fly.
- I guess this is happening because my thrust curve doesn't start at time == 0 seconds
- Therefore the SciPy's integrator adopts really large time steps, which make the thrust curve to be ignored.

**To Reproduce**

Try to follow the code block below. You're gonna see that the rocket never leaves the position (0, 0, 0).

```python
liquid_motor = LiquidMotor(
        thrust_source="data/SEBLM/test124_Thrust_Curve.csv",
        burn_time=(8, 20), # this is important to reproduce the error!
        dry_mass=10,
        dry_inertia=(5, 5, 0.2),
        center_of_dry_mass_position=0,
        nozzle_position=-1.364,
        nozzle_radius=0.069 / 2,
    )
liquid_motor.add_tank(pressurant_tank, position=2.007)
liquid_motor.add_tank(fuel_tank, position=-1.048)
liquid_motor.add_tank(oxidizer_tank, position=0.711)

calisto_liquid_modded= Rocket(
        radius=0.0635,
        mass=14.426,
        inertia=(6.321, 6.321, 0.034),
        power_off_drag="data/calisto/powerOffDragCurve.csv",
        power_on_drag="data/calisto/powerOnDragCurve.csv",
        center_of_mass_without_motor=0,
        coordinate_system_orientation="tail_to_nose",
    )
calisto_liquid_modded.add_motor(liquid_motor, position=-1.373)

test_flight = Flight(
        rocket=calisto_liquid_modded,
        environment=Environment(),
        rail_length=5,
        inclination=85,
        heading=0,
        max_time_step=np.inf # this is important to reproduce the error!
)

test_flight.all_info()
```

**Expected behavior**

A description of what you expected to happen.

**Screenshots**

If applicable, add screenshots to help explain your problem.

**Additional context**

```bash
tests/test_flight.py::test_liquid_motor_flight
  RocketPy\rocketpy\plots\flight_plots.py:107: UserWarning: Attempting to set identical low and high zlims makes transformation singular; automatically expanding.
    ax1.set_zlim3d([0, max_z])

tests/test_flight.py::test_liquid_motor_flight
  RocketPy\rocketpy\plots\flight_plots.py:108: UserWarning: Attempting to set identical low and high ylims makes transformation singular; automatically expanding.
    ax1.set_ylim3d([min_xy, max_xy])

tests/test_flight.py::test_liquid_motor_flight
  RocketPy\rocketpy\plots\flight_plots.py:109: UserWarning: Attempting to set identical low and high xlims makes transformation singular; automatically expanding.
    ax1.set_xlim3d([min_xy, max_xy])
```

**Possible solutions:**

While discussing this internally, we came up with some suggestions:
- We could set the `max_time_step` as 0.1 * `burn_duration`, for example, to ensure that any simulation will have at least 10 steps.
- The default value of `max_time_step` could now be set to 0.1 instead of `np.inf`
- We could stop initializing the flight simulations at t=0s, instead, starting it at t= burn_time[0]

Gui-FernandesBR, 2023-09-16:
I think it's not a good idea to start the simulation with a initial_time different of t=0.
Otherwise, we would lose some insights in the flight, for example, the "apogee_time" would become the flight time plus the time until ignition starts. This would be a bit confusing I guess.

It's better to adjust the thrust curve to always start at 0, probably.

I'm open to discussions anyway

williambrenner, 2023-09-16:
This bug happens to me frequently even when the thrust curve starts with 0. The way I fix it is the same as suggested (changing the maxTimeStep to 0.1) as shown in the screenshot

[image]

The simulations of this screenshot had a thrust curve that starts like this

[image]

In Monte Carlo simulations this also happens sometimes, usually when the standard deviation of the burn time is too high. To correct the dispersion analysis, I have to reduce the standard deviation.

Exsthain, 2025-02-14:
I've encountered the same error when trying to simulate a low-powered rocket, but this time, even after setting the maxTimeStep to 0.1, it only partially mitigated the issue, as I still encounter the following error:

"
File c:\Users\...\.conda\lib\site-packages\numpy\lib\function_base.py:630, in asarray_chkfinite(a, dtype, order)
    628 a = asarray(a, dtype=dtype, order=order)
    629 if a.dtype.char in typecodes['AllFloat'] and not np.isfinite(a).all():
--> 630     raise ValueError(
    631         "array must not contain infs or NaNs")
    632 return a

ValueError: array must not contain infs or NaNs
"

Additionally, the original issue of the rocket failing to leave the rail persists in Monte Carlo simulations, even with a very low standard deviation for all parameters.

Gui-FernandesBR, 2026-11-02:
Reopened: the fix is merged on `develop` and will close when v1.14.0 is released to `master`.