RocketPy-Team/RocketPy #437: Addition of Time Delay Parachute Deployment

AidanPowers, 2023-10-13:
**Is your feature request related to a problem? Please describe.**

I am simulating a [Loc-IV ](https://www.apogeerockets.com/Rocket_Kits/Skill_Level_3_Kits/LOC_IV) kit powered off of a [Aerotech 38mm HP SU DMS Motor - H100W-14A](https://www.apogeerockets.com/Rocket-Motors/AeroTech-Motors/38mm-Motors-Single-Use/Aerotech-38mm-HP-SU-Motor-H100W-14A)

This kit has no avionics, and instead uses a fixed time delay charge in order to deploy the parachute.

Currently RocketPy does not have the functionality to define the parachute deploy charge as a fixed time event.

This can be either a fixed time from burnout, or fixed time from launch. This depends on if the motor is a [core burner or end burner](https://www.rocketryforum.com/threads/when-does-the-time-start-for-delays.16575/post-145207). 

**Describe the solution you'd like**

I would like to be able to specify a time delay, here are the approaches I can come up with.
1) Adding T to the state vector given in Parachute.triggerfunc().
2) Adding key word support for flight phases, similar to apogee. IE: Launch then adding a lag.
3) Adding delay charge to motor characteristics.

**Additional context**

Here is the code base I am currently working to which this problem is relevant. 
https://github.com/AidanPowers/AAE-439-Traj

I am willing to contribute to a fix, but this would be my first time contributing to a project so I may need some assistance.

Gui-FernandesBR, 2024-08-19:
You could create a callable trigger function that always return True, and then use the `lag` argument to set the delay time. Something like described here: https://discord.com/channels/765037887016140840/1273822920501629000

Gui-FernandesBR, 2026-11-02:
Reopened: the fix is merged on `develop` and will close when v1.14.0 is released to `master`.