RocketPy-Team/RocketPy #443: Function not working properly if interpolation method is set for n-d arrays

brunosorban, 2023-10-27:
**Describe the bug**
If a `Function` object is created using, for example, a 3D array and the user sets the interpolation method different from `shepard`, it is accepting only 1 input when it should accept 2. There should be a protection so the user cannot define the interpolation method if a n-d array is given as input.

**To Reproduce**

[image]

**Expected behavior**

[image]

Also, the extrapolation method must be manually set for n-D arrays, meaning it is not optional.

[image]

Gui-FernandesBR, 2023-10-28:
@brunosorban did you consider this commentary?

https://github.com/RocketPy-Team/RocketPy/blob/0d0079718753fe7b3f9262b63271ddef4ac50e7e/rocketpy/mathutils/function.py#L54

Gui-FernandesBR, 2023-10-28:
Just for reference:

```python
>>> from rocketpy import function
>>> a = [(0,0,1), (1,1,2)]
>>> f = Function(a, ['x', 'y'], ['z'])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "RocketPy\rocketpy\mathutils\function.py", line 82, in __init__
    self.set_source(source)
  File "RocketPy\rocketpy\mathutils\function.py", line 237, in set_source
    self.set_interpolation("shepard")
  File "RocketPy\rocketpy\mathutils\function.py", line 290, in set_interpolation
    self.set_get_value_opt()
  File "RocketPy\rocketpy\mathutils\function.py", line 331, in set_get_value_opt
    raise ValueError(f"Invalid extrapolation type {self.__extrapolation__}")
ValueError: Invalid extrapolation type None
>>>
>>> f = Function(a, ['x', 'y'], ['z'], extrapolation="natural") 
>>> f(0,1)
1.5
```

- Ok, it's true that it throwing an error when you do not set the extrapolation type, this should be documented or fixed somehow.
- Idk why only `shepard` interpolation type is supported. I'd like to hear what are the limitations of the other types under n-D functions. Any comments on that @giovaniceotto @phmbressan ?

brunosorban, 2023-10-28:
> @brunosorban did you consider this commentary? It's a part of the official documentation.
> 
> https://github.com/RocketPy-Team/RocketPy/blob/0d0079718753fe7b3f9262b63271ddef4ac50e7e/rocketpy/mathutils/function.py#L54

Yep, it's how I found out how to make it work, but I spent a few minutes trying to figure out what was going on. I opened the issue because it would nice to have the proper checks to make it more user friendly. I volunteer to implement it's okay with you

Gui-FernandesBR, 2023-10-28:
> > @brunosorban did you consider this commentary? It's a part of the official documentation.
> > https://github.com/RocketPy-Team/RocketPy/blob/0d0079718753fe7b3f9262b63271ddef4ac50e7e/rocketpy/mathutils/function.py#L54
> 
> Yep, it's how I found out how to make it work, but I spent a few minutes trying to figure out what was going on. I opened the issue because it would nice to have the proper checks to make it more user friendly. I volunteer to implement it's okay with you.
> 
> About the interpolation, it's because Shepard is the only method suited for spatial interpolation currently implemented, which is the case for n-D distributions.

I'd love to hear your suggestion for such checks. We still struggle to find a good way of checking the user inputs.

I understand that Shepard is currently the only one implemented, but Idk why others were not implemented yet (and if they are really necessary since shepard seems to be good enough). We do have bilinear interpolation implemented in other part of the code: https://github.com/RocketPy-Team/RocketPy/blob/0d0079718753fe7b3f9262b63271ddef4ac50e7e/rocketpy/tools.py#L67

Gui-FernandesBR, 2023-10-28:
@brunosorban this part of the code is already checking the extrapolation type, 
code is dated from 4 years ago, so I think we could make something more robust already, what do you think?

https://github.com/RocketPy-Team/RocketPy/blob/0d0079718753fe7b3f9262b63271ddef4ac50e7e/rocketpy/mathutils/function.py#L325C1-L336C1

brunosorban, 2023-10-28:
> @brunosorban this part of the code is already checking the extrapolation type, code is dated from 4 years ago, so I think we could make something more robust already, what do you think? [...]

That would be great. And regarding the user checks, we could have a method that is called every time the user updates (or create) an `Function` object that verifies the input type and, if it is not `"shepard"` in the n-D case, updates it and raise a warning.

Gui-FernandesBR, 2023-10-28:
> That would be great. And regarding the user checks, we could have a method that is called every time the user updates (or create) an `Function` object that verifies the input type and, if it is not [...]

Could you suggest a method to validate the extrapolation and give an warning message in case of wrong ipnut? You can do that in a PR if you feel confortable

Gui-FernandesBR, 2023-11-19:
Solved! Thank you Bruno for point that out.

closed 2023-11-19 by Gui-FernandesBR