RocketPy-Team/RocketPy #447: TST: separate tests/conftest.py into smaller files

Gui-FernandesBR, 2023-10-29:
**Is your feature request related to a problem? Please describe.**

This was initially commented here: https://github.com/RocketPy-Team/RocketPy/pull/408#discussion_r1323807449

The idea is to have a better experience when reading the fixtures definitions! We could separate the files following the same module structure that in have in the rocketpy module

**Describe the solution you'd like**

* We could follow this: https://stackoverflow.com/a/65035367/19971645

**Additional context**

A further investigation is still needed

lucasfourier, 2023-11-23:
Gonna take a look into this while solving the following issue: https://github.com/RocketPy-Team/RocketPy/issues/480

lucasfourier, 2024-02-19:
Closing issue! Mission accomplished!

closed 2024-02-19 by lucasfourier