RocketPy-Team/RocketPy #480: TST: Restructuring the test suite

lucasfourier, 2023-11-23:
**Is your feature request related to a problem? Please describe.**

RocketPy test suite is currently very slow. Among other factors, this has to do with the fact that unit tests and integration tests (which are being treated as unit tests in many ways) are being kept in the same folder. 

**Describe the solution you'd like**

Restructuring the test suite will bring clarity as well as proper treatment depending on the nature of the test. With regards to the unit tests, they need to be refactored: commented, rewritten and etc.

**Additional context**

Even though some methods had tests written for them, bugs occurred. Of course, this situation might happen even though unit tests are available for a certain method. However, that also indicates that the test suite/testing process is not robust enough. As an example of this, check this bug:
https://github.com/RocketPy-Team/RocketPy/pull/476

Gui-FernandesBR, 2023-11-23:
Good idea! 

I liked the idea of separating unit and integration tests.

Here some other suggestions (not mandatory):

- https://pypi.org/project/pytest-random-order/
- https://github.com/kevlened/pytest-parallel

Also, the acceptance tests are quite important, we should improve them by bringing more examples of real flights to be tested.

lucasfourier, 2023-11-23:
Nice! Gonna check the suggestions!

lucasfourier, 2023-12-01:
------"Algorithm" that will run the reorganization:------

**1.** For each module x in the tests directory, copy x into the new directory unit tests.
**2.** After that, classify those tests which are certainly unit and leave them there. 
**3.** Remove the ones which are not unit and are inside the module. (Observe that they are not deleted forever, you could still find them in the original test_function.py testing module).
**4.** Now, having copied each testing module to the unit test folder, refactor every unit test in every testing module (if needed)
**5.** Finished unit testing organization. Start developing integration tests.

Pull request in each step.

closed 2024-07-04 by Gui-FernandesBR