RocketPy-Team/RocketPy #482: Calculations F*ck up without throwing error, when Fluid Volume not right

theDiego21, 2023-11-23:
I encountered a Bug, in which, when the Volume of a Fluid is greater than the Tank Volume, the engine gets calculated just fine, but when you add the engine to a rocket, the centre of gravity of the rocket is set to thousands of meters negative.

The Rocket therefore does not get drawn properly and the calculation of the Flight goes (i think) into an endless loop. 
(I tried letting it run for 15 mins, then i cancelled the calculation)

Sometimes, an error (Overfill or underfill) is thrown, but most of the times, the calculation of the engine does not throw an error
Environment used: Google Colab

phmbressan, 2023-11-23:
Thank you for reporting this Bug and creating an issue. 

This was already detected and is being fixed by the hotfix PR #479 . A release containing this hotfix will be released in a few days.

Additionally, could you provide more information about which `Tank` class produced this issue (or even a code sample) so we can check if the planned hotfix covers this case?

theDiego21, 2023-11-23:
The Tank used was a MassFlowRateBasedTank.
Here is the code sample, which caused the bug:

# Define fluids
oxidizer_liq = Fluid(name="LOX", density=1141)
fuel_liq = Fluid(name="ethanol", density=789)
pressurant_gas = Fluid(name="N2", density=1.034)

# Define tanks geometry
eth_tank_shape = CylindricalTank(radius = 13 / 100, height = 0.345409407550068, spherical_caps = True)
lox_tank_shape = CylindricalTank(radius = 13 / 100, height = 0.28827610851596, spherical_caps = True)
pressurant_tank_shape = CylindricalTank(radius = 6 / 100, height = 0.22, spherical_caps = True)

# Define tanks
oxidizer_tank = MassFlowRateBasedTank(
    name="oxidizer tank",
    geometry=eth_tank_shape,
    flux_time=5,
    initial_liquid_mass=4.695,
    initial_gas_mass=0,#.1880196263187668,
    liquid_mass_flow_rate_in=0,
    liquid_mass_flow_rate_out=lambda t: 0.4815384615384616, #/ 3 * exp(-0.25 * t),
    gas_mass_flow_rate_in=0,
    gas_mass_flow_rate_out=0,
    liquid=oxidizer_liq,
    gas=pressurant_gas,
)

fuel_tank = MassFlowRateBasedTank(
    name="fuel tank",
    geometry=lox_tank_shape,
    flux_time=5,
    initial_liquid_mass=3.611538461538462,
    initial_gas_mass=0,#.1880196263187668,
    liquid_mass_flow_rate_in=0,
    liquid_mass_flow_rate_out=lambda t: 0.626, #/ 3 * exp(-0.25 * t),
    gas_mass_flow_rate_in=0,
    gas_mass_flow_rate_out=0,
    liquid=fuel_liq,
    gas=pressurant_gas,
)

pressurant_tank = MassFlowRateBasedTank(
    name="pressurant tank",
    geometry=pressurant_tank_shape,
    flux_time=5,
    initial_liquid_mass=0,
    initial_gas_mass=0.1980196263187668,
    liquid_mass_flow_rate_in=0,
    liquid_mass_flow_rate_out=0,
    gas_mass_flow_rate_in=0,
    gas_mass_flow_rate_out=lambda t: 2.397258605729897/63,
    liquid=fuel_liq,
    gas=pressurant_gas,
)

example_liquid = LiquidMotor(
    thrust_source=lambda t: 2000,# - 100 * t**2,
    dry_mass=2,
    dry_inertia=(0.125, 0.125, 0.002),
    nozzle_radius=13 / 100,
    center_of_dry_mass_position=0.7,
    nozzle_position=0,
    burn_time=5,
    coordinate_system_orientation="nozzle_to_combustion_chamber",
)
example_liquid.add_tank(tank=oxidizer_tank, position=1.5)
example_liquid.add_tank(tank=fuel_tank, position=0.8)
example_liquid.add_tank(tank=pressurant_tank, position=1.1)
example_liquid.add_tank(tank=pressurant_tank, position=1.8)

example_liquid.draw()
example_liquid.info();

By adding the Motor to the Rocket using:
hedy.add_motor(example_liquid, position=-1.255)

The following output is produced, when hedy.draw() is called:
[image]

The Method hedy.info(); returns the following:
Inertia Details

Rocket Mass: 25.000 kg (without motor)
Rocket Dry Mass: 27.000 kg (with unloaded motor)
Rocket Loaded Mass: 35.703 kg (with loaded motor)
Rocket Inertia (with unloaded motor) 11: 7.016 kg*m2
Rocket Inertia (with unloaded motor) 22: 7.016 kg*m2
Rocket Inertia (with unloaded motor) 33: 0.036 kg*m2
Rocket Inertia (with unloaded motor) 12: 0.000 kg*m2
Rocket Inertia (with unloaded motor) 13: 0.000 kg*m2
Rocket Inertia (with unloaded motor) 23: 0.000 kg*m2

Geometrical Parameters

Rocket Maximum Radius: 0.13 m
Rocket Frontal Area: 0.053093 m2

Rocket Distances
Rocket Center of Dry Mass - Center of Mass without Motor: 0.041 m
Rocket Center of Dry Mass - Nozzle Exit: 1.214 m
Rocket Center of Dry Mass - Center of Propellant Mass: 8125060076944.635 m
Rocket Center of Mass - Rocket Loaded Center of Mass: 1980500324599.361 m

Aerodynamics Lift Coefficient Derivatives

Nose Cone Lift Coefficient Derivative: 2.000/rad
Fins Lift Coefficient Derivative: 11.071/rad
Tail Lift Coefficient Derivative: -0.490/rad

Center of Pressure

Nose Cone Center of Pressure position: 2.799 m
Fins Center of Pressure position: -0.906 m
Tail Center of Pressure position: -1.224 m

Stability

Center of Mass position (time=0): -1980500324599.402 m
Initial Static Margin (mach=0, time=0): -7617308940765.757 c
Final Static Margin (mach=0, time=burn_out): -9130882444823.043 c
Rocket Center of Mass (time=0) - Center of Pressure (mach=0): 1980500324599.097 m

Parachute Details

Parachute Main with a cd_s of 10.0000 m2
Ejection signal trigger: 800 m (AGL)
Ejection system refresh rate: 105.000 Hz
Time between ejection signal is triggered and the parachute is fully opened: 1.5 s

Parachute Details

Parachute Drogue with a cd_s of 1.0000 m2
Ejection signal trigger: At Apogee
Ejection system refresh rate: 105.000 Hz
Time between ejection signal is triggered and the parachute is fully opened: 1.5 s

closed 2023-12-07 by Gui-FernandesBR