RocketPy-Team/RocketPy #506: BUG: KeyError: 'I_11' and AttributeError: 'NoneType' object has no attribute 'reset'

Gui-FernandesBR, 2023-12-07:
**Describe the bug**

There's a problem on the Motor.I_11 method. The sum `propellant_I_11 + dry_I_11` is returning `None` for a strange reason. The order of the operations create a non sense result:

```python
        dry_I_11 += (
            self.dry_mass
            * (self.center_of_dry_mass_position - self.center_of_mass) ** 2
        )
```

This could be translated into: 

```python
        np.array +=(
            float 
            * (float - Function) **2
        )
```

**To Reproduce**

I have a code example with me but I'll still ask if I can share it. But try to create a liquid motor and then access its I_11 property, is it a function?

**Expected behavior**

Code should run without problems.

Actually, the #488 solves this problem. I've also tested to invert the operations. I think we can wait for the next release (2~3 weeks) to solve this.

**Screenshots**

[image]

**Additional context**

Issue firstly reported on discord by `Django [DARE]`.

The error seems to be present since version v1.0.0

phmbressan, 2023-12-08:
@Gui-FernandesBR could you run this code in the `develop` branch. I believe this error was fixed by the PR #488 that is already in `develop`.

Gui-FernandesBR, 2024-01-16:
The number 488 already solved this one.
A few weeks from now it should be available at master branch, but the problem is solved tho.