RocketPy-Team/RocketPy #529: ENH: adds `axial_acceleration` attribute to the Flight class

Gui-FernandesBR, 2024-01-12:
**Is your feature request related to a problem? Please describe.**

Some rockets may present an avionics system measuring the rocket axial acceleration instead of the "ground reference" acceleration.
RocketPy should be able to easily create

**Describe the solution you'd like**

- Needs to check how to transform the current acceleration components to the axial acceleration.
- A cached property should be created in the Flight class

emab123, 2024-02-28:
If the purpose of this feature is to determine the acceleration experienced by the avionics, in order to include the effects of angular acceleration, it would require a parameter for the location of the avionics package (specifically the accelerometer). Maybe have the Cg position as default and look for an Avionics component type?

Gui-FernandesBR, 2024-02-28:
> If the purpose of this feature is to determine the acceleration experienced by the avionics, in order to include the effects of angular acceleration, it would require a parameter for the location of [...]

Hello Eric, thanks for your comment.

I realized the description of this issue is really bad and does not reflect what I had in mind. I will update it asap.

What I meant was to simply transform the acceleration from one frame (earth) to other (rocket). That should be less complicated.

You do have a great point regarding the location of the avionics system. I think im the future we will be doing something like what you've suggested. There's the #sensors issue that is related to this future.

Gui-FernandesBR, 2025-11-19:
> Hello [@Gui-FernandesBR](https://github.com/Gui-FernandesBR) ,
> 
> I'm very interested in helping with this feature!
> 
> Just to confirm the scope of this issue: you are looking for the implementation of the linear acceleration of the Center of Mass (Cg) projected onto the rocket's axis of symmetry.
> 
> Witch would correspond to:
> 
> ```
>     self.ax * self.attitude_vector_x + self.ay * self.attitude_vector_y + self.az * self.attitude_vector_z
> ```
> 
> I can prepare a Pull Request with this implementation. Let me know if that sounds good!
> 
> Your follow-up comment about changing the frame of reference is not too clear. Could please elaborate on it?

@MateusStano is the best person to answer this one