RocketPy-Team/RocketPy #546: ENH: give me a scissor or apply `Function.clip` and `Function.crop`

Gui-FernandesBR, 2024-02-06:
**Is your feature request related to a problem? Please describe.**

Mutating Function objects is magical because sometimes we need to do some particular operations with them.
This issue requests two new methods to create new Function objects from other array-based Function objects.

**Describe the solution you'd like**

Please implement the following: 
- [ ] `Function.clip` : limits the `y_array` to a certain range, for instance, limit the sin function to be constrained between -1 an 1. This should delete any datapoint with the  y value out of the range.
- [ ] `Function.crop` : limits the `x_array` to a certain range, for instance, limit the domain of a R1->R1 function to be limited by x=0 and x=1. This should delete any datapoint with the x values out of the range. 

Some considerations:
- The methods should take the array source of the Function, remove the undesired data, return
- Once we remove data from the array and recreate a function object, all the interpolation coefficients will be recalculated. Should that be a problem?

I'm not sure how we would do (if possible) the same operations in non array-based Functions, so I'm leaving them out of the scope here.

**Additional context**

- [numpy.clip](https://numpy.org/doc/stable/reference/generated/numpy.clip.html)
- https://stackoverflow.com/questions/43463523/center-crop-a-numpy-array
- **This is already an implementation that can be used**: https://github.com/RocketPy-Team/RocketPy/blob/d552093a3be062d2deb3f18ff5ff89774a295d5d/rocketpy/Function.py#L630

Rickisterr, 2025-05-08:
Good day, I was wondering if this issue is still under consideration and if so, may I work on it?

Gui-FernandesBR, 2025-05-08:
> Good day, I was wondering if this issue is still under consideration and if so, may I work on it?

yes it is, but there's a lot already done here: https://github.com/RocketPy-Team/RocketPy/tree/enh/liq

Can you openup a PR?

Rickisterr, 2025-05-08:
understood; then I will refer to and use the linked branch's implementation and finish this feature up and submit a PR as soon as I finish

Also, for further clarification, the methods should also clip the functions themselves for when source is a callable lambda function as well, right? Not just when it has a set of discrete values that are to be clipped?

Thank you!

Gui-FernandesBR, 2025-05-08:
> understood; then I will refer to and use the linked branch's implementation and finish this feature up and submit a PR as soon as I finish
>
> Also, for further clarification, the methods should also [...]

yes they should.

Rickisterr, 2025-05-12:
Good day, I just wanted to notify that I have opened up a PR regarding this. Thank you so much!

Gui-FernandesBR, 2025-05-12:
> Good day, I just wanted to notify that I have opened up a PR regarding this. Thank you so much!

Amazing work, @Rickisterr !!

Rickisterr, 2025-05-12:
Thank you so much! I really appreciate all the help and being able to work on this and hope I can contribute even more!

Gui-FernandesBR, 2025-05-12:
> Thank you so much! I really appreciate all the help and being able to work on this and hope I can contribute even more!

If you want to pick another open issue to work on, please feel free to do so.
I believe your PR should be merged son, the review will probably be quite smooth.