RocketPy-Team/RocketPy #560: Incorrect/incomplete docstring in environment.py

chasgior214, 2024-02-21:
**Describe the bug**

Incorrect/incomplete documentation of the Environment class's \_\_init__ method.

**To Reproduce**

Open environment.py and go to the \_\_init__ method (also [here](https://docs.rocketpy.org/en/latest/_modules/rocketpy/environment/environment.html#Environment.__init__))

**Expected behavior**

Correct (no reference to launch rail length) and complete documentation.

Gui-FernandesBR, 2024-02-22:
Hello @chasgior214 , thank you for opening this issue.

I'm a bit lost tho, could you point me exactly what is wrong with the current documentation?
I don't see any reference to launch rail length and the documentation seems complete enough to me.

[image]

chasgior214, 2024-02-22:
Hi @Gui-FernandesBR 

The mention of launch rail length is at the top of the screenshot. There's also two incomplete bits. The first is the rest of that paragraph that has an unfinished sentence ("is loaded until another"). The second is an unfinished sentence in the description of the gravity parameter.

Gui-FernandesBR, 2024-02-22:
> Hi @Gui-FernandesBR
>
> The mention of launch rail length is at the top of the screenshot. There's also two incomplete bits. The first is the rest of that paragraph that has an unfinished sentence ("is [...]

Got it. Thanks for specifying it. We will work on that soon enough

Gui-FernandesBR, 2024-02-22:
@phmbressan assigned to you, boss.

closed 2024-04-19 by Gui-FernandesBR